opentrack/opentrack #1574: Opentrack receives data but for some reason it is not reaching the simulator.

passer3d, 2022-11-14:
I am using for Flight Simulator 2020 and DCS, AITrack + Opentrack. For some reason they have stopped working. Opentrack receives the data from AITrack and there is movement. But the camera in both simulators does not work. I have updated both programs and reinstalled. I have checked the ports and settings. I launch all apps with admin permissions. And yes, I have also restarted the PC. I'm out of ideas. Can you give me a hand? Thank you!

sthalik, 2022-11-14:
What's the output method that you've selected?

passer3d, 2022-11-14:
Freetrack 2.0
I've tried with mouse emulation, just for testing and it's not working.

[image]

sthalik, 2022-11-14:
In "select interface", select "use both".

passer3d, 2022-11-14:
> 
thanks Stanislaw,
I tried that before but still not working. 
Mouse emulation should be working as basic test but don't. It's like something is blocking Opentrack. (antivirus is disabled).

passer3d, 2022-11-14:
I don't know why, but it's working.  Maybe a Javascript update I just done?
Thanks for your quick response!

closed 2022-11-14 by passer3d

sthalik, 2022-11-14:
Changing 'use freetrack' to 'use both' only affects games that are started after opentrack has started tracking (not just starting opentrack).

Good to know it's fixed.