opentrack/opentrack #1576: Neuralnet track 30fps on ps3eye

kagavva, 2022-11-26:
Hello, driver doesn't matter, ive try every tracker. But in neuralnet ive got only ~30fps.

GO63-samara, 2022-11-27:
I confirm. Same. Regardless of the settings, always 35 FPS.
@DaWelter Any ideas?

[image]

sthalik, 2022-11-27:
> Inference: 27 ms

That's CPU usage.

$\frac{1000}{27} \approx 37$

GO63-samara, 2022-11-27:
Thanks, it's clearer now.
In my opinion, neurelnet tracker needs a readme about settings.

sthalik, 2022-11-27:
That screenshot looks real ugly. Seems like the Fusion theme didn't activate. What Windows version are you using?

GO63-samara, 2022-11-27:
This is Windows 7. What should the correct screenshot look like?

sthalik, 2022-11-27:
Scratch that. It's Fusion alright, just the color scheme is different.

A Windows 7 holdout, truly a prestigious gentleman [image]

DaWelter, 2022-11-28:
Yeah, @sthalik figured it out. Get a modern CPU :-) For comparison
i7-core2 (ca. 2014): 20 to 25 ms
Ryzen 9 5900X:  10 ms
:-)

guodong000, 2022-12-30:
try to increase "Thread Count"

beggehorn, 2023-12-01:
I'm getting only 45 FPS on a very fast PC (Intel 10980xe). I've assigned more cores and have also tried different USB3 ports (including directly to the back of the case). No changes...

beggehorn, 2023-12-14:
Update! Setting thread count in OpenTrack a bit higher than I tried before seems to work. A setting of "4" allows me to hit 60FPS using nuralnet.

NuclearPhoenixx, 2024-01-15:
Any way to increase the FPS to the hardware limit of the camera? It's a bummer that the settings won't allow you to use any more than 60 fps. There is only 30 FPS and 60 FPS and "Default" is the same 30 FPS. Any way to "hack" some config file to increase this or is this actually some limit with the face detection algorithm or something?

beggehorn, 2024-01-15:
I didn't try allocating more threads as the CPU hit becomes enormous quickly even on my 18 core Intel. I wonder if neuralnet could be optimized further?

NuclearPhoenixx, 2024-01-15:
Had like a ~15% load at solid 60 FPS with just a single allocated thread on my 12400F so no idea what's going on with your system. At least having the option to increase the FPS would be nice. If it doesn't run well you can always lower it, so I don't 100% understand it why there is such a low artificial limits except if neuralnet itself can't go any faster.

DaWelter, 2024-01-16:
> Any way to increase the FPS to the hardware limit of the camera?

Not sure if it's possible to query the camera for it's FPS. You need more than 60? I mean 60 is quite quick. But I think we could add a "no limit" setting.

> I wonder if neuralnet could be optimized further?

Possibly. There are two things I'd like to try eventually
* Running the neuralnet on the GPU
* Integer quantization of the neuralnet

We depend on the runtime framework (ONNX) to run the model. And AFAIK @sthalik compiled it with all the fancy cpu optimizations you can have.

sthalik, 2024-01-16:
Running it on the integrated Intel GPU might be worthwhile, in order to save cycles for the real GPU.

beggehorn, 2024-01-17:
> Had like a ~15% load at solid 60 FPS with just a single allocated thread on my 12400F so no idea what's going on with your system. At least having the option to increase the FPS would be nice. If it doesn't run well you can always lower it, so I don't fully understand why there is such a low artificial limits except maybe if neuralnet itself can't go any faster.

I played with it more today and actually found I can get a solid 60FPS with a single thread at about 8% CPU usage on my i9-10980xe. However, I'm using an app called Project Lasso that runs a core-managing profile whenever ProTools is launched (this is a digital audio workstation PC). As soon as I'm running ProTools, opentrack won't hit 60FPS until I've allocated four threads. I need to figure out how to optimize this as I will always be running opentrack while using ProTools,

sthalik, 2024-01-19:
I'm also using Process Lasso. If you're running opentrack on a single core then you can set it to realtime priority. But having all cores running with realtime priority at the same time will cause a Windows crash.

beggehorn, 2024-01-29:
> I'm also using Process Lasso. If you're running opentrack on a single core then you can set it to realtime priority. But having all cores running with realtime priority at the same time will cause a Windows crash.

Thanks. I'm having some issues getting the best performance with opentrack and Process Lasso configuratiion. Prior to this, I was mainly using PL to configure my DAW (ProTools) to only use cores 2-35 (hyperthreaded) in order to leave overhead for the OS on cores 0-1. 

I tried setting opentrack to "1 thread" in the opentrack options and then tried setting its CPU affinity to Core 1 (always) but I only get 35-40 fps this way.

Do you have any specific settings I can try. I have to admit, my experience with Process Lasso is very limited. I've been running it for a long time but had only configured it for my DAW (as descrivbed) and then have left it be.

I appreciate any help you might have to offer!

beggehorn, 2024-01-31:
> I'm also using Process Lasso. If you're running opentrack on a single core then you can set it to realtime priority. But having all cores running with realtime priority at the same time will cause a Windows crash.

Would you mind sharing your specific Process Lasso Settings for opentrack as well as what value you set the "thread count" for neuralnet? There are so many options, I'm having trouble finding any combination that works better than simply adding more threads in opentrack's neralnet settings.

Thx!!

sthalik, 2024-01-31:
Since you actually have a total of 35 hyperthreaded cores, my original advice regarding process priority is incorrect. Sorry.

beggehorn, 2024-01-31:
> Since you actually have a total of 35 hyperthreaded cores, my original advice regarding process priority is incorrect. Sorry.

In the interest of finding the most optimal way to run opentracker with neuralnet, I've tried disabling HT in my BIOS so I'm back to 18 cores. I'm still finding I only get 50-60 FPS (varies wildly) with 2 Threads allocated in opentrack. This is with no rules for opentrack set in PL and priority at "normal". When I bump opentrack up to "Real-Time" priority it doesn't seem to make much of a difference.

Would love to see how you've configured opentrack and PL.

OrbitingOcelot, 2024-04-13:
> Not sure if it's possible to query the camera for it's FPS. You need more than 60? I mean 60 is quite quick. But I think we could add a "no limit" setting.

Hi, just chiming in - a "no limit" would be really nice, as the PS3 camera can hit up to 120fps - I got one specifically because of this :)