opentrack/opentrack #1580: OT Wine output protocol and Proton/Wine (Steam Play)

gerrygk, 2022-12-01:
Hi all,

as mentioned in issue 1560, I now established the wine output for OT successfully but I can't start OT with Proton as the set wine variant in wine settings. This leads to the error 

> Error occurred while loading protocol Wine -- Windows layer for Unix
> Couldn't find a Steam runtime.

As I don't have an idea where this file is located, not to mention that I don't know how to fix this problem basically, any help is appreciated by now.

Thx in advance

cofilippo, 2022-12-03:
I get the same error after setting for wine and running it.

YosypC, 2024-03-30:
I have the same problem.
Weirdly, right after compilation with Wine, OT was able to find the Steam Runtime but not the App ID game currently ran.
Now that I have rebooted, OT doesn't even find Steam.

YosypC, 2024-04-10:
I've found that editing `opentrack/proto-wine/proton.cpp`'s `static const char* steam_paths[]`'s and `static const char* runtime_paths[]`'s element by appending the home tilde `~` in the string, the program successfully recognizes the Steam Runtime.
Which doesn't make sense since 

```
for (const char* path : runtime_paths) {
        QDir dir(QDir::homePath() + path);
        if (dir.exists())
            runtime_path = dir.absolutePath();
    }
```
[QDir::homePath()](https://doc.qt.io/qt-6/qdir.html#homePath) should provide one by itself.

Moreover, it doesn't find the AppIDs, still.

YosypC, 2024-04-10:
I've just realized that, at least under Linux Mint, the ~/.steam directory only contains so:

```
bin
bin32
bin64
debian-installation
registry.vdf
root
sdk32
sdk64
steam
steam.pid
steam.pipe
steam.token
```

Meanwhile the runtime paths are listed as so:

```
static const char* runtime_paths[] = {
    "/.local/share/Steam/ubuntu12_32/steam-runtime",
    "/.steam/ubuntu12_32/steam-runtime",
};
```

which, in this case, is clearly wrong. I have very limited knowledge on... everything, so I can't really do much.