opentrack/opentrack #1595: Opentrack crashes when opening options or input options when using PointTracker

ericesminombre, 2023-01-02:
OpenTrack freezes when opening options or input options when using PointTracker. I've tried 2022.3.2, 2022.3.1 and 2.3.13, and the same thing happens, although in 2.3.13 it doesen't crash when entering options. I used opentrack 2022.3.2 the day before writting this and it worked normally. This issue also occurs when using other inputs such as EasyTracker or WiiPointTracker. Output or filter don't seam to matter.

ericesminombre, 2023-01-02:
I found the problem. It seems that the usb drivers ¡(for samsung phones) I installed to use my phone as a webcam are incomaptible with opentrack.

closed 2023-01-02 by ericesminombre

sthalik, 2023-01-03:
Is this still occuring if you delete `opentrack-video-ps3eye.dll` as a workaround?