opentrack/opentrack #1608: Request to add graphics card processing and hardware acceleration capabilities

huzesama, 2023-02-03:
Request to add graphics card processing and hardware acceleration capabilities

sthalik, 2023-02-03:
Are you sure you've got the right project?

huzesama, 2023-02-03:
Yes, I hope opentrack can use Core Graphics card which is not commonly used to share the pressure of calculation. Actually, I don't know if it's an under-configuration problem, because I've been running into a lot of problems with the Eurotruck Simulator 2, and even stopped working. (The above is translated through the translation software, do not know whether the meaning is wrong)

sthalik, 2023-02-03:
The neuralnet tracker is the only one that could benefit from it. For the rest that's too much copying from and to VRAM. What do you think, @DaWelter? Could be as simple as enabling selection of the ONNX backend.