opentrack/opentrack #1610: Opentrack not receiving any input on UDP

t348575, 2023-02-03:
I am using an stm32 + IMU + compass to calculate x,y,z and euler angles, and sending as IEEE LE format to my pc using an esp8266-01, running opentrack with the input as "UDP over network" any my port 4242.

I am following stuff told in these issues:
* [747](https://github.com/opentrack/opentrack/issues/747)
* [1398](https://github.com/opentrack/opentrack/issues/1398)
* [1203](https://github.com/opentrack/opentrack/issues/1203)

I have confirmed that my pc is able to get the data. If i use a sample program to receive udp packets, i am able to see correct data being received. Only opentrack is not showing anything. Windows firewall shows opentrack is enabled. I tried some other ports also, none worked. I have only one connected network interface (no wsl, or vm networks).

What could be causing this issue?

Note: I am sending data updates every 1ms.

opentrack is definitely listening:
[image]

Sample program to verify my PC is getting data:
```rs
use std::net::UdpSocket;

#[derive(Debug, PartialEq, Copy, Clone)]
pub struct MovementData {
    pub x: f64,
    pub y: f64,
    pub z: f64,
    pub yaw: f64,
    pub pitch: f64,
    pub roll: f64,
}

impl MovementData {
    fn from(data: &[u8]) -> Self {
        Self {
            x: f64::from_le_bytes(data[0..8].try_into().unwrap()),
            y: f64::from_le_bytes(data[8..16].try_into().unwrap()),
            z: f64::from_le_bytes(data[16..24].try_into().unwrap()),
            yaw: f64::from_le_bytes(data[24..32].try_into().unwrap()),
            pitch: f64::from_le_bytes(data[32..40].try_into().unwrap()),
            roll: f64::from_le_bytes(data[40..48].try_into().unwrap()),
        }
    }
}

fn main() {
    let socket = UdpSocket::bind("192.168.0.10:4242").unwrap();
    let mut buf = [0; 480000];
    let (amt, src) = socket.recv_from(&mut buf).unwrap();
    loop {
        let buf = &mut buf[..amt];
        for item in buf.chunks(48) {
            if item.len() == 48 {
                println!("{:?}", MovementData::from(item));
            }
        }
    }
}
```

Output:
[image]

t348575, 2023-02-04:
Ok I am getting data in opentrack now... any idea why my data is garbled like this? If this is irrelevant you can close the issue...
[image]

I'm using an mpu6050 with an hmc5883L, and using [this](https://github.com/jmagnuson/ahrs-rs) to do sensor fusion using Madgwick filters. If I multiply my output euler angles with `180.0 / PI` this is what I am getting... (It is sitting still on my table)

My code:
```rs
use std::f64::consts::PI;
use std::net::UdpSocket;
use std::time::Duration;

use ahrs::Ahrs;
use ahrs::Madgwick;
use nalgebra::Vector3;
use serde::Serialize;

#[derive(Debug, PartialEq, Copy, Clone, Serialize)]
pub struct MovementData {
    pub x: f64,
    pub y: f64,
    pub z: f64,
    pub yaw: f64,
    pub pitch: f64,
    pub roll: f64,
}

type Vec3 = Vector3<f64>;

#[derive(Debug, PartialEq, Copy, Clone)]
pub struct AccelGyro {
    pub accel: Vec3,
    pub gyro: Vec3,
}

impl AccelGyro {
    fn from(data: &[u8]) -> Self {
        AccelGyro {
            accel: Vector3::new(
                i16::from_le_bytes(data[0..2].try_into().unwrap()) as f64,
                i16::from_le_bytes(data[2..4].try_into().unwrap()) as f64,
                i16::from_le_bytes(data[4..6].try_into().unwrap()) as f64,
            ),
            gyro: Vector3::new(
                i16::from_le_bytes(data[8..10].try_into().unwrap()) as f64,
                i16::from_le_bytes(data[10..12].try_into().unwrap()) as f64,
                i16::from_le_bytes(data[12..14].try_into().unwrap()) as f64,
            )
        }
    }
}

fn get_mag(data: &[u8]) -> Vec3 {
    Vector3::new(
        i16::from_be_bytes(data[0..2].try_into().unwrap()) as f64,
        i16::from_be_bytes(data[4..6].try_into().unwrap()) as f64,
        i16::from_be_bytes(data[2..4].try_into().unwrap()) as f64,
    )
}

const TO_RAD: f64 = PI / 180.0;
const TO_PI: f64 = 180.0 / PI;

fn main() {
    let socket = UdpSocket::bind("0.0.0.0:4242").unwrap();
    socket.set_read_timeout(Some(Duration::from_millis(5))).unwrap();

    let mut buf = [0u8; 4096];
    let mut imu = Madgwick::new(0.00309, 0.001); // consistent latency measured from controller is 3.09ms
    loop {
        let recv = socket.recv_from(&mut buf);
        if recv.is_err() {
            continue;
        }

        let (n, _) = recv.unwrap();
        for item in buf[..n].chunks_exact(20) {
            let accel_gyro = AccelGyro::from(&item[0..14]);
            let mag = get_mag(&item[14..]);
            let quat = imu.update(&(accel_gyro.gyro * TO_RAD), &accel_gyro.accel, &mag);
            if quat.is_err() {
                continue;
            }

            let quat = quat.unwrap();
            let (roll, pitch, yaw) = quat.euler_angles();
            let coords = quat.coords;
            let data = MovementData {
                x: coords[0],
                y: coords[1],
                z: coords[2],
                yaw: yaw * TO_PI,
                pitch: pitch * TO_PI,
                roll :roll * TO_PI,
            };
            socket.send_to(&bincode::serialize(&data).unwrap(), "192.168.0.10:4243");
            println!("{:?}", data);
        }
    }
}
```

GO63-samara, 2023-02-04:
All three sensors G,A,M require calibration.
The weakest point in the tracker is the use of a magnetometer. His work is affected by close stationary metal objects. Such as a chair headrest frame etc.
First, try replacing the chair with a stool and calibrate the magnetometer.
If it doesn't help, temporarily turn off the magnetometer, so you can find out the reason. Madgwick has a tracker option without a magnetometer.

If interested, here is my version of the tracker, not Madgwick, but Mahoney (Razor AHRS), In Russian:
https://sites.google.com/site/diyheadtracking/home/5dof-tracker
Your SensorVariant 2 (GY-85):
https://sites.google.com/site/diyheadtracking/home/5dof-tracker/sensors-units

By the way, I heard that `hmc5883L` has already been discontinued, `qmc5883` are now used instead. Maybe this is the reason. Then your sensor option is SensorVariant 9 (GY-85).

GO63-samara, 2023-02-04:
Something I hastened. I want to clarify.
`mpu6050 + hmc5883L` is not `GY85`. It's more like `GY521 + GY271` or `GY521 + GY273`.
Then your SensorVariant is 7, 11, 12 or 13.

In the screenshot, the yaw and roll angles are close to 90 degrees. This may be due to the fact that you did not take into account the mutual orientation of the axes of the sensors G, A, M.
Good luck!

t348575, 2023-02-05:
Thanks for the info, closing because there is no issue with opentrack.

closed 2023-02-05 by t348575