opentrack/opentrack #1612: Error occurred while loading tracker TrackHat Point Tracker - Failed to open camera

Blubbfisch, 2023-02-10:
MSI Z170A GAMING PRO CARBON (MS-7A12)
Intel Core i5-6600K
Windows 11 Pro 22H2
Trackhat Opentrack 2.2b
Opentrack 2022.3.2
----------------------------
Thinkpad T460s
Intel Core i5-6300U
Windows 11 Pro 22H2
Trackhat Opentrack 2.2b
----------------------------

Hello,
it's shameful that I am looking for help here, but the support from trackhat does not seem to exist.

When I klick the Trackhat Opentrack start button using the first system, I do get the error message: "Error occurred while loading tracker TrackHat Point Tracker - Failed to open camera 'TrackHat Sensor'" respectively "Error occurred while loading tracker TrackHat Point Tracker - Failed to open camera '' using Opentrack 2022.3.2

I haven't found any way to fix this issue so far. I tried differnt USB-Ports, port reset, different drivers etc. 
Interestingly, the sensor is recognized as such by the system.
Searching the internet for help wasn't successful as well. I had the same issue with Windows 10 Pro, too.

Using the Thinkpad I have no issues at all.

Can anybody help me with this issue?

Thanks and regards,

Marek

charlyselo, 2023-02-13:
Same for me with Windows 10, app version is 2.2b 
PS3 camera and TrackHat clip plus works fine, tested with CL-Eye Test app. Opentruck just didn't recognice camera.

sthalik, 2023-05-06:
The TrackHat tracker is for the failed V1 sensor. It was a device doing light point detection in hardware with only limited release because it had a 96x96 sensor and we couldn't get it to work. Use the PT tracker instead.

sthalik, 2023-05-06:
When you open Skype or Discord or a similar program that has a list of cameras, what is the name of the camera you want to connect to?

svencc, 2023-05-06:
As far as I remember the ps3 camera was not visible in the list of the cameras. Only my other webcam appeared. I have read that the ps3 eye camera driver I used is replaced by another one. This was the old one which was suggested by trackhat a few years ago. It seems that this does not work anymore.

I discarded the use of this ps3camera and use the latest version of opentrack in combination of AITrack which works pretty awesome with my Razor Kyio webcam!

sthalik, 2023-05-06:
Try this then. https://github.com/opentrack/opentrack/wiki/PS3-Eye-open-driver-instructions

Blubbfisch, 2023-05-07:
My issue is resolved. I had to change the mainboard and with the new mainboard Opentrack  is working as expected.

Since I opend this thread, do I have to close it now, or can I leave it open, since there are others with the same issue?

sthalik, 2023-05-07:
Let's leave it open for a few months. People don't like to talk in closed issues.

DavidMM0501, 2023-09-22:
I have just upgraded my PC MB and Processor, and installed Windows 11. I am now having the same issue as others on this thread. I followed all the steps in the link given by sthalik on May 7 successfully (well, I got no errors) and then tried trackhat (same versoin 2.2b) but I got the SAME error message - failed to open camera etc. I did not check Device Manager then, instead I tried CL-Eye Test again - which worked. When I looked at the properties of the camera in Device Manager, the driver shown there was Code Laboratories, not the open source one at the link above. Did trying CL-Eye Test cause it to install its own driver again? Either way trackhat did not work as mentioned above, so are there any other ideas on how to resolve the issue?
Thanks for keeping the thread open.

DavidMM0501, 2023-09-22:
Hi, I added a comment earlier, but do not see it here. I want to add more, but since I may not have posted it, I will start again.
I have just upgraded my PC motherboard and processor, and moved to Windows 11. I am having the same issues that were the reason for this thread to be opened. I have now followed the exact steps in the link provided by sthalik on May 7. The first time I did not check Device Manager after the driver install - but trackhat gave the same error (Failed to open Track Hat Sensor. I then tried CL-Eye Test which worked. I then checked Device Manager and found the driver to be that of CL - so it must have reinstalled it. the second time round I checked Device Manager first, and it does NOT list the camera. Needless to say trackhat gives the same error. When I clicked on finish install it did so, but I also got another window about selecting a filter - this is not mentioned in the link - and the device was not listed, well not in a format I could recognise. 
Has anybody any more advice or ideas for me to try?
Thanks for keeping the thread open.

DavidMM0501, 2023-09-22:
Two additional points - I did reboot after the 2nd attempt at the driver installation, before checking Device Manager. And I do not have an option to select Camera Settings under Tracker in trackhat, as shown in step 5 in the link.

DavidMM0501, 2023-10-05:
To close this up, sthalik kindly helped me resolve the issues off line. After trying to download the drivers a third time, they were successfully installed. My main issue was I was still using the Trackhat opentrack, app instead of the opentrack tracker app. All works fine now.