opentrack/opentrack #1614: in game jittery

anakinsolo, 2023-02-14:
It is very jittery in game. It feels like everything moves by itself when I am in game, but when I alt-tab to the opentrack window, it stops.....So I switch the game to windows mode, set the windows size as small as opentrack windows and observe this:
- When I am in focus of the game windows, the jittery happens. The camera starts to jump and look around by itself. The reading from the game data suggests some heavy and fast movement
- When I am in focus of the opentrack windows, the game runs in background, every is smooth. No jittery no jumping around.

Game used: Microsoft Flight Simulator 2020
Input: Neuralnet tracker
Output: freetrack 2.0 enhanced
Filter: Accela
[image]

Here is neuralnet settings. I have played with different resolutions, FOV or even FPS. I also recalibrated many times, or increased the thread counts. None of them helped.
[image]

Here is the freetrack settings. I have played with different interface. That doesn't really matter
[image]

Accela settings are basic. Output only Pitch and Yaw. Smooth and Deadzone to the max.
[image]

WvandeHoef, 2023-02-19:
I am having the same issue since today. Using opentrack-2.3.12 with Delanclip and PS3Eye Cam for almost 3 years now without issues, but today am noticing I cannot fly MSFS 2020 with TrackIR enabled, as the image is constantly twitching. Behaviour is also visible in Opentrack when watching the Octopus.

Could this be related to a Windows Update? I'm still on Win 10 Pro 21H2 / 19044.2486

When checking the PointTacker Status it shows the FPS jumping from 1 FPS to 60 FPS, where I did set it to 25 FPS

WvandeHoef, 2023-02-19:
I was able to solve the problem by saving the profiles, removing my old version, downloading the latest version 2022.3.2 and install that one. Now the twitching is completely gone and the movement is fluid as before.

As input I have PointTracker 1.1

CegaMega, 2023-02-27:
I have jittery issues too, tried removing it and reinstalling but didn't work as expected. Digital Combat Simulator

Pieloth, 2023-02-27:
As part of the jittery issue experienced with Neuralnet tracker, I've noticed by looking Windows ressource monitor, that there seems to be some kind of memory leak with the trackers, and mostly visible with Neuralnet tracker, each time you stop and start tracking
As a workaround for my own usage, I stop Opentrack and restart it each time I'm using it, and this reduces significantly the jitters

But I guess we have some kind of memory leaks underneath, that impacts performances also (jittery), after series of tracking start/stop

anakinsolo, 2023-04-26:
I use hamilton and it helps a bit, mainly because the jitters do not happen as heavily as the accela. Still if I want to hold my head still to do some stuffs like configuring MFD, that's not very accurate. I used to have TrackIR and now I also have VR headset so I could tell that Opentrack does have some drifting issue

sutex, 2023-05-12:
@anakinsolo I'm using the PS3 eye camera and tried Hamilton, and the jitters have gone for the most part depending on the level of smoothing and max distance applied, yet the tracking is noticeably slower and tracks worse than when using Accela. 

I've tried different settings, limited as they are, but as yet nothing has worked to my satisfaction. I like the idea of a Neuralnet tracker, but coming from Delan Clip, the former still needs some work.

anakinsolo, 2023-05-12:
> @anakinsolo I'm using the PS3 eye camera and tried Hamilton, and the jitters have gone for the most part depending on the level of smoothing and max distance applied, yet the tracking is noticeably [...]

Sorry to hear, I do not use PS3 eye cam but a Logitech C920. For the most part, it has been more stable than any of these options
- accela + neuranet
- accela + trackAI
But it still drifts in-game quite a bit, especially when I zoom into a MFD and need to keep my head still to configure something.

I have not used Delan Clip, only TrackIR back then, but it was a long time ago so I do not remember the difference