opentrack/opentrack #1617: Can't get past 15 Fps no matter what settings I use

ZipperKopf, 2023-02-24:
[image]

As I said no matter what settings I use I can't get the webcam to go over 15 fps and I don't understand why. The tracking is smooth but if I'm only getting 15fps I'm using half of the webcams 30 fps output. I have tried putting the res at 640x480 and that does nothing. 

Any explanation/ tips would be greatly appreciated.

GO63-samara, 2023-02-24:
https://github.com/opentrack/opentrack/issues/1576

ZipperKopf, 2023-02-24:
> 

This thread doesn't give me a solution at all.

sthalik, 2023-02-24:
Exposure too high.

ZipperKopf, 2023-02-24:
> Exposure too high or MJPEG not enabled.

I tried changing both and nothing happened, still 15fps

sthalik, 2023-02-24:
I had the same camera model (Logitech C525) and it ran in 15 FPS when a setting called 'RightLight' was enabled.

ZipperKopf, 2023-02-24:
> I had the same camera model (Logitech C525) and it ran in 15 FPS when a setting called 'RightLight' was enabled.

This fixed it! Thank you so much!

closed 2023-02-24 by ZipperKopf