opentrack/opentrack #1618: RFC: proto-wine UI changes

Priton-CE, 2023-02-25:
On [my fork of opentrack](https://github.com/Priton-CE/opentrack-StarCitizen) for Star Citizen the SC LUG and I developed an improved UI for selecting wine versions other than system wine.
This was required since essentially all games running in a launcher like Lutris are not using system wine but instead a packed wine version (referred to as "runner").

If this is something that would be useful for opentrack itself we would be happy to provide a PR without the other changes of the opentrack-StarCitizen fork.

[image]
[image]

sthalik, 2023-02-26:
These are good changes, except for `gameid2` handling. It exists so that `NPClient.dll` [knows](https://github.com/opentrack/opentrack/blob/master/contrib/npclient/npclient.c) that opentrack agrees with its `gameid`. Because of [TrackIR Enhanced](https://naturalpointofview.blogspot.com/p/trackir-anticompetition.html) titles, data can't be sent until that's the case.

So drop the thunk making changes to `wine::pose()` (this also solves allocating a string every time `pose()` is called) as well as defaulting `gameid` to `-1` in the header.

In `wine::initialize()` there's an indentation error.

Other than that, the Wine protocol in its current state has suffered due to neglect, I think your changes should be merged immediately and any issues can be solved later.

Priton-CE, 2023-02-26:
I think I resolved the issues you are talking about. The `QString gamename` was already there when I forked but it should only allocate once, when the agreement is taking place.

Things I didn't change:
1. The CMake change that enables wine by default.
2. The change to the registry creation in the wrapper. No idea if that will cause changes with other games.

Priton-CE, 2023-02-26:
created a PR https://github.com/opentrack/opentrack/pull/1619

closed 2023-02-26 by Priton-CE