opentrack/opentrack #1626: Tracker only working when OpenTrack window in foreground / focused

arithex, 2023-03-12:
[Possibly same/similar issue as https://github.com/opentrack/opentrack/issues/1499 but that was fixed.]

Installed v2022.3.2 on latest Windows 10, 22H2

Using joystick input source (thumbstick on gamepad) it appears tracking only responds to thumbstick input while the OpenTrack UI window is active/foreground.

Click on the desktop background -- tracking stops.  Click back on OpenTrack window -- tracking resumes.

arithex, 2023-03-12:
Also, the toggle commands appear to be borked.  Toggle-while-held works as expected, but ordinary "toggle" does not.  Seems to break tracking until I explicitly "stop tracking" and then restart tracking.

Not sure if these problems are specific to joystick input or not.. seems like a fundamental problem that would affect everything, but nobody else is complaining, so maybe nobody is using gamepad thumbstick this way?

karageee, 2023-05-16:
yes, I'm also having this problem. 

Installed v2022.3.2 on latest Windows 11 22H2 as well

Although the software does run and track my head movement albeit with extreme lag. I think it's more of window's issue rather than the program, but I can't seem to find the fix for windows.

oneup03, 2024-06-14:
Still an issue with joystick input on windows 11 at least.