opentrack/opentrack #1633: [REGRESSION] OT 2023.1.0 wine-wrapper process fails to start (crashes)?

XenoPL, 2023-03-26:
OpenTrack 2023.1.0  wine-wrapper process crashes at start. 
Program self compiled under Fedora 37 64-bit
cmake config:
`-DCMAKE_BUILD_TYPE=Release -DSDK_XPLANE="$xplane_sdk_dir" -DSDK_ARUCO_LIBPATH="$arucolib" -DONNXRuntime_DIR="$onnxruntime_dir" -DSDK_WINE=ON -DCMAKE_INSTALL_PREFIX="$otdir" -DCMAKE_CXX_FLAGS="-fopenmp"`
where dir variables point to aruco lib dir and ONNX/Xplane SDK dirs. 
Wine version : 8.4-stagging (official Fedora repo)

Tested with PointTracker and NeuralNet, using accela filter and wine  output config:
 - system wine
 - valid prefix set
 - ESYNC: ON FSYNC:
 - Protocol: Both
Tracking in OT window works fine, however wine apps (game: Falcon4 AF and TrackIR client included in OT docs see no headtracking), no opentrack-wrapper-wine.exe.so present while tracking is on.

Same config used by OT 2022.3.2 installed side by side (compiled with the same settings/packages version) works just fine.

erlking, 2023-03-30:
after you run it (and start tracking), can you check the registry, following paths:
Software\Freetrack\FreeTrackClient
Software\NaturalPoint\NATURALPOINT\NPClient Location

maybe it points to the old 2022.3.2 install ?

i also heard that sometimes people have to re-create their prefix for whatever reason.

XenoPL, 2023-04-07:
It seems OT 2023.1.0 won't work with 32bit wine prefixes. I've made two fresh ones for a test, it fails with 32bit prefix but works just fine with 64bit one.