opentrack/opentrack #1634: neuralnet tracker pitching upwards on yaw

ber8749, 2023-03-26:
Hello,

I've been using the neuralnet tracker for several months now, and while it has generally performed quite well, I have been encountering a positioning issue that I can't quite resolve. Basically, when I turn my head left or right (yaw) the head tracker pitches up. Even after calibrating and centering the tracker, this issue persists. 

Just for context, my camera is sitting on top of my monitor, and angled downwards towards my head, so I'm not looking directly at the camera, but rather, slightly below it.

I've fiddled with the settings quite a bit, but have yet to find a suitable resolution for this problem. Did I miss a setting that would resolve this issue?

[image]

[image]

[image]

Pieloth, 2023-03-26:
Hello,

I've checked on my side, I get the same kind of behavior between pitch and yaw (although I don't use yaw angles as wide as you do), but I noticed that this behavior was the same with previous version 2022.3.2 (with the settings I used for that version)

I did not care too much about it, assuming it was due to how the webcam was seeing my face, and then deducing the 6 DOF parameters

But maybe there's something to check here with the ONNX lib outputs?

GO63-samara, 2023-03-26:
Try changing the centering method:

`Options` - `Shortcuts` - `Centering method` - `Point`

ber8749, 2023-03-27:
@GO63-samara Thanks for the suggestion, but already currently use `Point` as the centering method

ber8749, 2023-04-03:
Thank you for the response @DaWelter 

In my case, I believe this issue is rooted in the fact that my face is not directly aligned with the camera, but rather slightly offset below the camera (i.e. my camera is sitting on my monitor angled down towards my face).

I assume moving my camera so it is below me angled upwards would have the opposite effect: the tracker would pitch downwards on yaw.

Would it be possible to allow the end user to dampen the pitch at the extremes of yaw based on the offset angle of their camera? Example, by default the offset is set to `0`. Positive values dampen negative pitch. Negative values dampen positive pitch.

Thoughts?

DaWelter, 2023-04-25:
So, @ber8749 , does this "Wireless VR 360" option help?

ber8749, 2023-04-26:
Sorry, @DaWelter, I misunderstood your earlier comment.

Unfortunately, the method you describe (look at middle of monitor and hit "center" button) does not dampen the unwanted pitch when turning my head.

DaWelter, 2023-04-27:
Just to be 100% clear ...
[image]
Is set, right? Then I'm out of ideas.

ber8749, 2023-04-27:
Oh okay, I didn't realize "Wireless VR 360" was a centering option, my mistake.
So, I just tried using it, and unfortunately it did not seem to dampen the unwanted pitch at the extremes of yaw.

Cat-Lady, 2024-02-01:
Just wanted to mention that it is still the problem. It doesn't matter if neuralnet or other method (tracker marker, for example) is used, so it seems it is, indeed, problem with opentrack centering calculation.

lowpass777, 2024-05-05:
I'm wondering if anyone did find a solution/ workaround? I'm having the same issue.

Gokip1, 2024-06-12:
I'm having the same issue, everything I have tried nothing seems to be working. Has anyone found a solution for it?

Cat-Lady, 2024-06-13:
Sadly, the only thing that I can suggest to somehow mitigate the issue, is to place camera *above* the screen/head plane, not below (if possible). The issue still exist in both cases, but is *much* more pronounced if the camera is looking up at you.

I hope proper fix will get integrated at some point :(