opentrack/opentrack #1649: opentrack 2023.1.0

Rdtcmttsa, 2023-05-02:
### Discussed in https://github.com/opentrack/opentrack/discussions/1630

<div type='discussions-op-text'>

<sup>Originally posted by **sthalik** March 21, 2023</sup>
### Compatibility

This release build is only available as a 64-bit executable. This is because the SDK for the new tracker `tracker/beam` is temporarily only available as an amd64 binary.

Only CPUs supporting AVX will be able to run it. Please provide feedback on compatibiltiy and performance compared to older releases.

### Changes

- `tracker/beam`: new tracker for the by @eyedav (Davide Mameli) for the [Eyeware Beam](https://beam.eyeware.tech/).
- `tracker/neuralnet`: tracking quality improved by @DaWelter (see 08f1fcad1c74e25f97641a0ccbd229b267ec528c for details).
- `tracker/neuralnet`: Russian translation by @Perlovka.
- `video/ps3eye`: experimentally enable retry of bulk data transfers for flaky USB controllers.
- `tracker/realsense`: downgrade to SDK 2016 R2.
- `tracker/pt`, `tracker/aruco`: fix tracking failing to start by not being able to select a camera when only one is available.
- `proto/wine`: improvements to Linux head tracking #1620 by @Priton-CE and #1616 by @fsimuser9430.

<hr /><em>This discussion was created from the release <a href='https://github.com/opentrack/opentrack/releases/tag/'>opentrack 2023.1.0</a>.</em></div>

Rdtcmttsa, 2023-05-02:
Hello folks im in the process of getting started on setting up my Tracking, I have downloaded the Opentrack software and the software for the PS3 camera, Im just needing to know how to install the OT software.
I have unzipped it, i just need to know which file goes where and how to install it.
Thanks....sorry for my ignorants.....RobertC

Len62, 2023-05-03:
Create a folder and name it as you like, unzip the files into that folder. There is no installation, just place the folder somewhere to your liking and create a shortcut to the exe. You can place the folder anywhere you like, I have a habit of just leaving it in my downloads folder.

luzpaz, 2023-08-25:
Issues should probably be closed.

closed 2023-11-05 by sthalik