opentrack/opentrack #1652: opentrack 2023.1.0

peter-mason-dpm, 2023-05-07:
### Discussed in https://github.com/opentrack/opentrack/discussions/1630

<div type='discussions-op-text'>

<sup>Originally posted by **sthalik** March 21, 2023</sup>
### Compatibility

This release build is only available as a 64-bit executable. This is because the SDK for the new tracker `tracker/beam` is temporarily only available as an amd64 binary.

Only CPUs supporting AVX will be able to run it. Please provide feedback on compatibiltiy and performance compared to older releases.

### Changes

- `tracker/beam`: new tracker for the by @eyedav (Davide Mameli) for the [Eyeware Beam](https://beam.eyeware.tech/).
- `tracker/neuralnet`: tracking quality improved by @DaWelter (see 08f1fcad1c74e25f97641a0ccbd229b267ec528c for details).
- `tracker/neuralnet`: Russian translation by @Perlovka.
- `video/ps3eye`: experimentally enable retry of bulk data transfers for flaky USB controllers.
- `tracker/realsense`: downgrade to SDK 2016 R2.
- `tracker/pt`, `tracker/aruco`: fix tracking failing to start by not being able to select a camera when only one is available.
- `proto/wine`: improvements to Linux head tracking #1620 by @Priton-CE and #1616 by @fsimuser9430.

<hr /><em>This discussion was created from the release <a href='https://github.com/opentrack/opentrack/releases/tag/'>opentrack 2023.1.0</a>.</em></div>

peter-mason-dpm, 2023-05-07:
I Cannot get opentrack 2023.1.0 to connect to X-Plane 11 0r 12.
I have a moving octopus and it ends there. I have tried all the ip addresses on offer from x-plane. I have tried enable TrackIR and xcamera nothing works
Other people say it just works but not for me.
Does anyone have an idea how to get it working?

sthalik, 2023-05-07:
Use the freetrack output method in opentrack. Enable TrackIR/TrackHat in X-Plane.

peter-mason-dpm, 2023-05-07:
Thank you for that. It works. All I need to do now is get the mapping stabilised but that should be okay.
Thanks
Peter

closed 2023-05-07 by peter-mason-dpm