opentrack/opentrack #1660: Opentrack not able to open webcam

apprehensivetarget, 2023-05-21:
I am trying to head track using a USB webcam and the Aruco trackers, but when I press start, my webcam light turns on for a brief instant and then turns off again and opentrack just displays a grey screen. I have tried passing the webcam feed through OBS first, as it is detected there, with the OBS virtual webcam module, but this does not work either, opentrack just displays a black screen this time, with I think an fps display, in the top left corner, in green, showing 1 fps. I have also tried disabling anti-virus, swapping USB ports and using the built-in webcam in my laptop, but to no avail. Any help would be greatly appreciated.

cultab, 2023-06-03:
> I have tried passing the webcam feed through OBS first, as it is detected there, with the OBS virtual webcam module, but this does not work either, opentrack just displays a black screen this time, [...]

I can have the exact same issue. I can provide more details if needed.

CodyAsano, 2023-07-04:
Same issue as well.

sthalik, 2023-07-04:
Try the 32-bit build first: https://ananke.misaki.pl/opentrack/opentrack-test-20230527_01.7z

CodyAsano, 2023-07-04:
Using this build fixed my issue! Thank you good Sir.

closed 2023-07-04 by sthalik