opentrack/opentrack #1661: OpenTrack Crash on Startup

wfurney13, 2023-05-21:
Reference #1212 and #522, experiencing a similar issue with crashing on startup for multiple versions of opentrack. I have previously had Logitech software on the computer that has been removed. The process crashes with an exception code c0000409. I receive the same error with previous versions of OpenTrack as well:

```
Microsoft (R) Windows Debugger Version 10.0.25200.1003 X86
Copyright (c) Microsoft Corporation. All rights reserved.

Loading Dump File [C:\Users\wfurn\AppData\Local\CrashDumps\opentrack.exe.22488.dmp]
User Mini Dump File: Only registers, stack and portions of memory are available

************* Path validation summary **************
Response                         Time (ms)     Location
Deferred                                       srv*
Symbol search path is: srv*
Executable search path is: 
Windows 10 Version 22621 MP (24 procs) Free x86 compatible
Product: WinNt, suite: SingleUserTS
Edition build lab: 22621.1.amd64fre.ni_release.220506-1250
Machine Name:
Debug session time: Sun May 21 13:15:13.000 2023 (UTC - 5:00)
System Uptime: 0 days 1:12:39.441
Process Uptime: 0 days 0:00:01.000
..............................................................
This dump file has an exception of interest stored in it.
The stored exception information can be accessed via .ecxr.
(57d8.6988): Security check failure or stack buffer overrun - code c0000409 (first/second chance not available)
Subcode: 0x5 FAST_FAIL_INVALID_ARG 
For analysis of this file, run !analyze -v
eax=00000001 ebx=0035b000 ecx=00000005 edx=000001f8 esi=00000000 edi=0019fc1c
eip=70cfbe66 esp=0019fbdc ebp=0019fbfc iopl=0         nv up ei pl nz na po nc
cs=0023  ss=002b  ds=002b  es=002b  fs=0053  gs=002b             efl=00000202
opentrack_user_interface!process_detector_worker::qt_metacall+0x6d36:
70cfbe66 cd29            int     29h
```

WinDbg Results:

```
Microsoft (R) Windows Debugger Version 10.0.25200.1003 X86
Copyright (c) Microsoft Corporation. All rights reserved.

CommandLine: C:\Users\wfurn\install\opentrack.exe

************* Path validation summary **************
Response                         Time (ms)     Location
Deferred                                       srv*
Symbol search path is: srv*
Executable search path is: 
ModLoad: 00ba0000 00bf0000   opentrack.exe
eax=00bae191 ebx=00f85000 ecx=00000000 edx=00000000 esi=00000000 edi=00000000
eip=77728800 esp=010ffa6c ebp=00000000 iopl=0         nv up ei pl nz na po nc
cs=0023  ss=002b  ds=002b  es=002b  fs=0053  gs=002b             efl=00000202
77728800 833de8f47d7700  cmp     dword ptr ds:[777DF4E8h],0 ds:002b:777df4e8=00000000
0:000> g
ModLoad: 776b0000 7785f000   ntdll.dll
ModLoad: 762d0000 763c0000   C:\WINDOWS\SysWOW64\KERNEL32.DLL
ModLoad: 75c20000 75e92000   C:\WINDOWS\SysWOW64\KERNELBASE.dll
ModLoad: 582b0000 582de000   C:\Users\wfurn\install\opentrack-pose-widget.dll
ModLoad: 58230000 58256000   C:\Users\wfurn\install\opentrack-migration.dll
ModLoad: 582e0000 58336000   C:\Users\wfurn\install\opentrack-user-interface.dll
ModLoad: 58260000 582ab000   C:\Users\wfurn\install\opentrack-logic.dll
ModLoad: 581e0000 58227000   C:\Users\wfurn\install\opentrack-options.dll
ModLoad: 58190000 581d4000   C:\Users\wfurn\install\opentrack-compat.dll
ModLoad: 57d00000 58184000   C:\Users\wfurn\install\Qt5Widgets.dll
ModLoad: 76d20000 7738d000   C:\WINDOWS\SysWOW64\SHELL32.dll
ModLoad: 759e0000 75b86000   C:\WINDOWS\SysWOW64\USER32.dll
ModLoad: 57720000 57cf3000   C:\Users\wfurn\install\Qt5Gui.dll
ModLoad: 75f70000 75fe9000   C:\WINDOWS\SysWOW64\msvcp_win.dll
ModLoad: 75ff0000 7600a000   C:\WINDOWS\SysWOW64\win32u.dll
ModLoad: 76c00000 76d12000   C:\WINDOWS\SysWOW64\ucrtbase.dll
ModLoad: 75420000 75443000   C:\WINDOWS\SysWOW64\GDI32.dll
ModLoad: 76470000 765bd000   C:\WINDOWS\SysWOW64\ole32.dll
ModLoad: 77440000 77522000   C:\WINDOWS\SysWOW64\gdi32full.dll
ModLoad: 75680000 758fb000   C:\WINDOWS\SysWOW64\combase.dll
ModLoad: 763c0000 7645c000   C:\WINDOWS\SysWOW64\OLEAUT32.dll
ModLoad: 75450000 7550a000   C:\WINDOWS\SysWOW64\RPCRT4.dll
ModLoad: 017c0000 0187a000   C:\WINDOWS\SysWOW64\RPCRT4.dll
ModLoad: 57190000 57717000   C:\Users\wfurn\install\Qt5Core.dll
Unload module C:\WINDOWS\SysWOW64\RPCRT4.dll at 017c0000
eax=40000036 ebx=00800000 ecx=00000000 edx=00000000 esi=c000022d edi=01455458
eip=777263bc esp=010fe87c ebp=010fe8b0 iopl=0         nv up ei pl nz na po nc
cs=0023  ss=002b  ds=002b  es=002b  fs=0053  gs=002b             efl=00000202
ntdll!NtUnmapViewOfSection+0xc:
777263bc c20800          ret     8
0:000> g
ModLoad: 57130000 5715b000   C:\Users\wfurn\install\opentrack-dinput.dll
ModLoad: 57160000 57188000   C:\Users\wfurn\install\opentrack-spline.dll
ModLoad: 75510000 7556f000   C:\WINDOWS\SysWOW64\WS2_32.dll
ModLoad: 57110000 5712c000   C:\Users\wfurn\install\opentrack-api.dll
ModLoad: 77620000 7769c000   C:\WINDOWS\SysWOW64\ADVAPI32.dll
ModLoad: 75ea0000 75f64000   C:\WINDOWS\SysWOW64\msvcrt.dll
ModLoad: 75b90000 75c12000   C:\WINDOWS\SysWOW64\sechost.dll
ModLoad: 74b40000 74b71000   C:\WINDOWS\SysWOW64\WINMM.dll
ModLoad: 717b0000 7182f000   C:\WINDOWS\SysWOW64\UxTheme.dll
ModLoad: 64100000 64124000   C:\WINDOWS\SysWOW64\dwmapi.dll
ModLoad: 66ce0000 66f17000   C:\WINDOWS\SysWOW64\d3d11.dll
ModLoad: 71fb0000 7207a000   C:\WINDOWS\SysWOW64\dxgi.dll
ModLoad: 62730000 62810000   C:\WINDOWS\SysWOW64\OPENGL32.dll
ModLoad: 71770000 71789000   C:\WINDOWS\SysWOW64\MPR.dll
ModLoad: 74950000 7496f000   C:\WINDOWS\SysWOW64\USERENV.dll
ModLoad: 748d0000 748d8000   C:\WINDOWS\SysWOW64\VERSION.dll
ModLoad: 718f0000 71904000   C:\WINDOWS\SysWOW64\NETAPI32.dll
ModLoad: 6e470000 6e4a8000   C:\WINDOWS\SysWOW64\DINPUT8.dll
ModLoad: 626e0000 62722000   C:\WINDOWS\SysWOW64\GLU32.dll
ModLoad: 720d0000 720db000   C:\WINDOWS\SysWOW64\NETUTILS.DLL
ModLoad: 701e0000 7020d000   C:\WINDOWS\SysWOW64\dxcore.dll
ModLoad: 71c00000 71c1d000   C:\WINDOWS\SysWOW64\SRVCLI.DLL
ModLoad: 747e0000 747eb000   C:\WINDOWS\SysWOW64\CRYPTBASE.DLL
(6004.639c): Break instruction exception - code 80000003 (first chance)
eax=00000000 ebx=00000000 ecx=93ac0000 edx=00000000 esi=01444288 edi=00f85000
eip=77767567 esp=010ff274 ebp=010ff2a0 iopl=0         nv up ei pl zr na pe nc
cs=0023  ss=002b  ds=002b  es=002b  fs=0053  gs=002b             efl=00000246
ntdll!LdrpDoDebuggerBreak+0x2b:
77767567 cc              int     3
0:000> g
ModLoad: 76bd0000 76bf5000   C:\WINDOWS\SysWOW64\IMM32.DLL
ModLoad: 6e320000 6e465000   C:\WINDOWS\SysWOW64\inputhost.dll
ModLoad: 6e250000 6e31d000   C:\WINDOWS\SysWOW64\CoreMessaging.dll
ModLoad: 760f0000 76152000   C:\WINDOWS\SysWOW64\bcryptPrimitives.dll
ModLoad: 56fb0000 57110000   C:\Users\wfurn\install\platforms\qwindows.dll
ModLoad: 72540000 7254f000   C:\WINDOWS\SysWOW64\WTSAPI32.dll
ModLoad: 74bd0000 74be3000   C:\WINDOWS\SysWOW64\kernel.appcore.dll
ModLoad: 765c0000 76680000   C:\WINDOWS\SysWOW64\SHCore.dll
ModLoad: 74ce0000 753ae000   C:\WINDOWS\SysWOW64\windows.storage.dll
ModLoad: 74c10000 74cd7000   C:\WINDOWS\SysWOW64\wintypes.dll
ModLoad: 760a0000 760eb000   C:\WINDOWS\SysWOW64\shlwapi.dll
ModLoad: 74bf0000 74c09000   C:\WINDOWS\SysWOW64\profapi.dll
ModLoad: 70d40000 70d86000   C:\WINDOWS\SysWOW64\powrprof.dll
ModLoad: 70ca0000 70cae000   C:\WINDOWS\SysWOW64\UMPDC.dll
(6004.639c): Security check failure or stack buffer overrun - code c0000409 (!!! second chance !!!)
Subcode: 0x5 FAST_FAIL_INVALID_ARG 
eax=00000001 ebx=00f85000 ecx=00000005 edx=000001f8 esi=00000000 edi=010ff820
eip=582fb8b6 esp=010ff7e0 ebp=010ff800 iopl=0         nv up ei pl nz na po nc
cs=0023  ss=002b  ds=002b  es=002b  fs=0053  gs=002b             efl=00000202
opentrack_user_interface!process_detector_worker::qt_metacall+0x6d46:
582fb8b6 cd29            int     29h
0:000> g
WARNING: Continuing a non-continuable exception
(6004.639c): Security check failure or stack buffer overrun - code c0000409 (!!! second chance !!!)
Subcode: 0x5 FAST_FAIL_INVALID_ARG 
eax=00000001 ebx=00f85000 ecx=00000005 edx=000001f8 esi=00000000 edi=010ff820
eip=582fb8b6 esp=010ff7e0 ebp=010ff800 iopl=0         nv up ei pl nz na po nc
cs=0023  ss=002b  ds=002b  es=002b  fs=0053  gs=002b             efl=00000202
opentrack_user_interface!process_detector_worker::qt_metacall+0x6d46:
582fb8b6 cd29            int     29h
```

Looks like the exception is thrown when loading UMPDC.dll. I wasn't able to find any other references to this particular issue. Does anyone have any suggestions for resolving this?

wfurney13, 2023-05-21:
[opentrackMiniDump134452123.zip](https://github.com/opentrack/opentrack/files/11525309/opentrackMiniDump134452123.zip)

sthalik, 2023-05-25:
I'm having trouble loading symbols for this minidump. Is this the version 2023.1.0? Since you're already using windbg, you can provide a backtrace on your own, using the win32-dbginfo package for the appropriate opentrack release.

> exception is thrown when loading UMPDC.dll.

This library is used internally by Windows Error Reporting, so this is after a crash already happened.

wfurney13, 2023-05-27:
Thanks for the info. Your help allowed me to remediate this issue using the win32-dbginfo package. 

Long story short: My PATH environment variable is greater than 2047 characters, which causes a stack buffer overrun exception that crashes the application. So the resolution is to shorten the path variable [like so.](https://stackoverflow.com/questions/34491244/environment-variable-is-too-large-on-windows-10) Once the path variable is short enough so that it does not overrun when OpenTrack tries to add to it, the application opens successfully. Maybe worth handling this in the future or warning the user that the PATH variable is too large? Here's the details on how I got there if anyone is interested.

Loading the trace file (with the symbols from the win32-dbginfo package) and running "analyze -v". There are now references to "invalid_parameter" around a strcat_s function call. Looking in the source code, strcat_s is used one time in init.cpp in the method add_win32_path. That is also referenced in the stack. So the parameter passed to strcat_s (env_path) is invalid. My environment path variable is overflowing, which is why we see the c0000409 exception for stack buffer overrun.

```

*******************************************************************************
*                                                                             *
*                        Exception Analysis                                   *
*                                                                             *
*******************************************************************************
............

EXCEPTION_RECORD:  (.exr -1)
ExceptionAddress: 00007ffacc9f3174 (opentrack_user_interface!_invoke_watson+0x0000000000000018)
   ExceptionCode: c0000409 (Security check failure or stack buffer overrun)
  ExceptionFlags: 00000001
NumberParameters: 1
   Parameter[0]: 0000000000000005
Subcode: 0x5 FAST_FAIL_INVALID_ARG 

FAULTING_THREAD:  00002444

PROCESS_NAME:  opentrack.exe

ERROR_CODE: (NTSTATUS) 0xc0000409 - The system detected an overrun of a stack-based buffer in this application. This overrun could potentially allow a malicious user to gain control of this application.

EXCEPTION_CODE_STR:  c0000409

EXCEPTION_PARAMETER1:  0000000000000005

STACK_TEXT:  
000000f1`d52ff3c0 00007ffa`cc9f313a     : 00000000`00000000 00000000`00000000 00000000`00000000 0000025e`00000000 : opentrack_user_interface!_invoke_watson+0x18
000000f1`d52ff3f0 00007ffa`cc9f3022     : 00000000`00000022 000000f1`d52ff4a0 00000000`00000002 0000025e`04681300 : opentrack_user_interface!_invalid_parameter_internal+0xce
000000f1`d52ff430 00007ffa`cc9f3155     : 00000000`00000022 00007ffa`cc9f4909 00000000`00000000 00000000`00000002 : opentrack_user_interface!_invalid_parameter+0x52
000000f1`d52ff4b0 00007ffa`cc9f2c71     : 00007ffa`cca0f0c8 00000000`00000002 0000025e`04622695 00000000`00000002 : opentrack_user_interface!_invalid_parameter_noinfo+0x19
000000f1`d52ff4f0 00007ffa`cc9d17c0     : 0000025e`045db8d0 0000025e`00000000 000000f1`d52ff5a0 00000000`00000002 : opentrack_user_interface!strcat_s+0x2d
000000f1`d52ff520 00007ffa`cc9d1a07     : 0000025e`0467fc90 0000025e`0467cc30 0000025e`0467f320 0000025e`0467f2d0 : opentrack_user_interface!add_win32_path+0x180
000000f1`d52ff5c0 00007ff7`1f45ddb9     : 0000025e`00000001 0000025e`045c6778 00000000`00001fc0 00007ff7`1f494180 : opentrack_user_interface!otr_main+0x107
000000f1`d52ff7a0 00007ff7`1f45f676     : 00000000`0000000a 00007ff7`1f45f6ed 00000000`00000000 00000000`00000000 : opentrack!WinMain+0x39
000000f1`d52ff810 00007ffb`7d7226ad     : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : opentrack!__scrt_common_main_seh+0x106
000000f1`d52ff850 00007ffb`7ee6a9f8     : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : KERNEL32!BaseThreadInitThunk+0x1d
000000f1`d52ff880 00000000`00000000     : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : ntdll!RtlUserThreadStart+0x28

STACK_COMMAND:  ~0s ; .cxr ; kb

FAULTING_SOURCE_LINE:  minkernel\crts\ucrt\src\appcrt\misc\invalid_parameter.cpp

FAULTING_SOURCE_FILE:  minkernel\crts\ucrt\src\appcrt\misc\invalid_parameter.cpp

FAULTING_SOURCE_LINE_NUMBER:  237

FAULTING_SOURCE_CODE:  
No source found for 'minkernel\crts\ucrt\src\appcrt\misc\invalid_parameter.cpp'

```

sthalik, 2023-05-27:
Amazing. Thanks for diagnosing it. I'll be releasing a test build soon. Can you still reproduce the bug if you set the PATH to what it originally was?

wfurney13, 2023-05-27:
Yes I can, I'll keep and eye out for the new build and let you know the difference

sthalik, 2023-05-27:
Try this: https://ananke.misaki.pl/opentrack/opentrack-test-20230527_01.7z

The debug info is inside and this is a 32-bit build.

wfurney13, 2023-05-28:
Yeah it works as expected. I forced my PATH variable back to >2047 characters and then reproduced the original issue with the current release. Confirmed that this test build works with PATH variable >2047 characters.

sthalik, 2023-05-28:
That's good to hear. Thanks for confirming it.

commit 71dbcd16 (2023-05-28): gui/init: fix crash on startup

Fixes: #1661
Reported, diagnosed and tested by: @wfurney13
closed 2023-05-28 by sthalik