opentrack/opentrack #1662: Cannot build on Linux with support for Wine

wilhem, 2023-05-26:
I'm following this tutorial found on this [link](https://volatilemark.github.io/opentrack-wine-guide/) step by step.

I'm trying to build Opentrack with X-Plane (Wine) support, because otherwise I cannot use it in my X-Plane 12 running on Ubuntu 22.04.

I installed all the necessary wine libraries but I stuck at this point:

```
-- User settings file '/home/ubuntu/X-Plane/opentrack/source/sdk-paths-davide@GNU-Linux.cmake' doesn't exist
-- module api
-- module compat
-- module csv
-- module cv
-- module filter-accela
-- module filter-ewma
-- module filter-hamilton
-- module filter-kalman
-- module user-interface
-- module logic
-- module migration
-- module executable
-- module options
-- module pose-widget
-- module proto-fgfs
-- module proto-libevdev
-- module proto-udp
-- module proto-wine
-- module qxt-mini
-- module spline
-- module tracker-easy
-- module tracker-freepie-udp
-- module tracker-fusion
-- module tracker-hatire
-- module tracker-linux-joystick
-- module tracker-pt-base
-- module tracker-pt
-- module tracker-s2bot
-- module tracker-test
-- module tracker-udp
-- module video-opencv
-- module video
Compile flags:
   CXX: -ggdb -Wall -Wextra -Wpedantic
   CXX_RELEASE: -O3 -march=native
Link flags:
   LINK_SHARED: 
   LINK_SHARED_RELEASE: 
Static archive flags:
   STATIC: 
   STATIC_RELEASE: 
--
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubuntu/X-Plane/opentrack/build
Consolidate compiler generated dependencies of target opentrack-version
Consolidate compiler generated dependencies of target opentrack-api
Consolidate compiler generated dependencies of target opentrack-qxt-mini
Consolidate compiler generated dependencies of target opentrack-compat
[  0%] Built target opentrack-version
[  1%] Built target opentrack-api
[  4%] Built target opentrack-qxt-mini
[ 19%] Built target opentrack-compat
[ 19%] Built target i18n-module-api
[ 19%] Built target i18n-module-qxt-mini
Consolidate compiler generated dependencies of target opentrack-options
[ 19%] Built target i18n-module-compat
[ 20%] Generating opentrack-wrapper-wine.exe.so
[ 28%] Built target opentrack-options
[ 30%] Generating moc_csv.cpp
[ 30%] Generating moc_camera.cpp
[ 30%] Generating moc_spline.cpp
[... 91 lines cut ...]
./build_script.sh: Zeile 21: Footer: command not found
```

Here is my system:

```
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=22.04
DISTRIB_CODENAME=jammy
DISTRIB_DESCRIPTION="Ubuntu 22.04.2 LTS"
PRETTY_NAME="Ubuntu 22.04.2 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.2 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy
```

Any help is super very appreciated

sthalik, 2023-05-26:
```console
$ git grep -- -m32 proto-wine
```

Edit that file. Replace `-m32` with `-m64`. 

When using this workaround, your `proto-wine` will only work on 64-bit wineprefixes.

sthalik, 2023-05-26:
The real solution is to build Wine on your own because the official Ubuntu version is badly packaged with respect to winegcc.

wilhem, 2023-05-26:
Thank you very much for your reply.

With a fresh downloaded repository I did exactly what you suggested.
Sadly it didn't work

```
ubuntu:~/X-Plane/opentrack/build$ make install -j$(nproc)
Consolidate compiler generated dependencies of target opentrack-version
Consolidate compiler generated dependencies of target opentrack-api
Consolidate compiler generated dependencies of target opentrack-qxt-mini
Consolidate compiler generated dependencies of target opentrack-compat
[  0%] Built target opentrack-version
[  1%] Built target opentrack-api
[  4%] Built target opentrack-qxt-mini
[ 19%] Built target opentrack-compat
[ 19%] Built target i18n-module-api
[ 19%] Built target i18n-module-qxt-mini
Consolidate compiler generated dependencies of target opentrack-options
[ 20%] Built target i18n-module-compat
[ 20%] Generating opentrack-wrapper-wine.exe.so
[ 28%] Built target opentrack-options
[ 28%] Generating qrc_posewidget.cpp
[ 28%] Generating moc_csv.cpp
[ 28%] Generating moc_export.cpp
[ 28%] Generating moc_camera.cpp
[ 29%] Generating moc_video-widget.cpp
[ 29%] Generating moc_spline.cpp
[ 29%] Generating ui_ftnoir_kalman_filtercontrols.h
[ 30%] Generating ui_ftnoir_hamilton_filtercontrols.h
[ 30%] Generating ui_ftnoir_ewma_filtercontrols.h
[ 30%] Generating ui_ftnoir_fgcontrols.h
[ 30%] Generating moc_kalman.cpp
[ 30%] Generating moc_ftnoir_filter_hamilton.cpp
[ 30%] Generating moc_ftnoir_filter_ewma2.cpp
[ 30%] Generating moc_axis-opts.cpp
[ 30%] Generating moc_export.cpp
[ 30%] Generating ui_ftnoir_libevdev_controls.h
[ 30%] Generating moc_pose-widget.cpp
[ 30%] Running lupdate for options
[ 31%] Building CXX object csv/CMakeFiles/opentrack-csv.dir/csv.cpp.o
[ 32%] Building CXX object video/CMakeFiles/opentrack-video.dir/camera.cpp.o
[ 32%] Generating moc_ftnoir_protocol_fg.cpp
[ 32%] Building CXX object video/CMakeFiles/opentrack-video.dir/video-widget.cpp.o
[ 32%] Generating moc_export.cpp
[ 32%] Generating moc_hamilton-tools.cpp
[ 32%] Generating moc_ftnoir_protocol_libevdev.cpp
[ 32%] Building CXX object filter-kalman/CMakeFiles/opentrack-filter-kalman.dir/kalman.cpp.o
[ 33%] Building CXX object pose-widget/CMakeFiles/opentrack-pose-widget.dir/pose-widget.cpp.o
[ 33%] Generating qrc_fg-protocol.cpp
[ 33%] Building CXX object filter-ewma2/CMakeFiles/opentrack-filter-ewma.dir/ftnoir_filter_ewma2.cpp.o
[ 33%] Built target i18n-module-options
[ 34%] Generating qrc_libevdev-protocol.cpp
[ 35%] Generating moc_spline-widget.cpp
[ 36%] Building CXX object filter-ewma2/CMakeFiles/opentrack-filter-ewma.dir/ftnoir_filter_ewma2_dialog.cpp.o
[ 36%] Building CXX object filter-hamilton/CMakeFiles/opentrack-filter-hamilton.dir/ftnoir_filter_hamilton.cpp.o
[ 36%] Building CXX object proto-flightgear/CMakeFiles/opentrack-proto-fgfs.dir/ftnoir_protocol_fg.cpp.o
[ 36%] Building CXX object proto-libevdev/CMakeFiles/opentrack-proto-libevdev.dir/ftnoir_protocol_libevdev.cpp.o
[ 36%] Building CXX object spline/CMakeFiles/opentrack-spline.dir/axis-opts.cpp.o
/usr/bin/ld: relocatable linking with relocations from format elf32-i386 (/usr/lib/i386-linux-gnu/wine/libwinecrt0.a(exe_entry.o)) to format elf64-x86-64 (opentrack-wrapper-wine.hjqIK4.o) is not supported
**winebuild: /usr/bin/ld failed with status 1
winegcc: /usr/lib/wine/winebuild failed
make[2]: *** [proto-wine/CMakeFiles/wine-wrapper.dir/build.make:75: proto-wine/opentrack-wrapper-wine.exe.so] Error 2
make[1]: *** [CMakeFiles/Makefile2:2351: proto-wine/CMakeFiles/wine-wrapper.dir/all] Error 2
make[1]: *** waiting for still running processes …**
[ 36%] Building CXX object filter-hamilton/CMakeFiles/opentrack-filter-hamilton.dir/ftnoir_filter_hamilton_dialog.cpp.o
[ 36%] Building CXX object video/CMakeFiles/opentrack-video.dir/moc_camera.cpp.o
[ 36%] Building CXX object video/CMakeFiles/opentrack-video.dir/moc_export.cpp.o
[ 37%] Building CXX object video/CMakeFiles/opentrack-video.dir/moc_video-widget.cpp.o
[ 37%] Building CXX object filter-hamilton/CMakeFiles/opentrack-filter-hamilton.dir/hamilton-tools.cpp.o
[ 37%] Building CXX object csv/CMakeFiles/opentrack-csv.dir/moc_csv.cpp.o
[... 16 lines cut ...]
[ 39%] Building CXX object proto-libevdev/CMakeFiles/opentrack-proto-libevdev.dir/moc_ftnoir_protocol_libevdev.cpp.o
[ 39%] Building CXX object spline/CMakeFiles/opentrack-spline.dir/spline-widget.cpp.o
[ 39%] Building CXX object proto-libevdev/CMakeFiles/opentrack-proto-libevdev.dir/qrc_libevdev-protocol.cpp.o
[ 39%] Building CXX object spline/CMakeFiles/opentrack-spline.dir/spline.cpp.o
[ 40%] Building CXX object spline/CMakeFiles/opentrack-spline.dir/moc_axis-opts.cpp.o
[ 40%] Building CXX object spline/CMakeFiles/opentrack-spline.dir/moc_export.cpp.o
[ 40%] Building CXX object spline/CMakeFiles/opentrack-spline.dir/moc_spline-widget.cpp.o
[ 40%] Building CXX object spline/CMakeFiles/opentrack-spline.dir/moc_spline.cpp.o
[ 41%] Building CXX object filter-kalman/CMakeFiles/opentrack-filter-kalman.dir/moc_kalman.cpp.o
[ 42%] Linking CXX shared library opentrack-proto-libevdev.so
[ 42%] Linking CXX shared library opentrack-proto-fgfs.so
[ 42%] Built target opentrack-proto-libevdev
[ 42%] Built target opentrack-proto-fgfs
[ 42%] Linking CXX shared library opentrack-filter-hamilton.so
[ 42%] Linking CXX shared library opentrack-filter-ewma.so
[ 42%] Built target opentrack-filter-hamilton
[ 42%] Built target opentrack-filter-ewma
[ 42%] Linking CXX shared library opentrack-spline.so
[ 42%] Built target opentrack-spline
[ 42%] Linking CXX shared library opentrack-filter-kalman.so
[ 42%] Built target opentrack-filter-kalman
make: *** [Makefile:136: all] Error 2
```

Should I build Wine directly?

wilhem, 2023-05-27:
Nope....

I built Wine from source (stable release) in /opt/wine without any error.

but I get the same error:

```
[  0%] Generating moc_plugin-api.cpp
[  0%] Generating moc_qxtglobal.cpp
[  0%] Generating moc_x11-keymap.cpp
[  0%] Generating moc_plugin-support.cpp
[  0%] Generating moc_variance.cpp
[  0%] Generating moc_export.cpp
[  0%] Generating moc_activation-context.cpp
[  1%] Generating moc_arch.cpp
[  1%] Generating moc_powerset.cpp
[  2%] Generating moc_qxtglobalshortcut.cpp
[  2%] Generating moc_qxtglobalshortcut_p.cpp
[  2%] Building CXX object CMakeFiles/opentrack-version.dir/version.cpp.o
[  2%] Generating moc_check-visible.cpp
[  2%] Generating moc_camera-names.cpp
[  2%] Generating moc_base-path.cpp
[  3%] Generating moc_enum-operators.cpp
[  3%] Generating moc_correlation-calibrator.cpp
[  3%] Generating moc_export.cpp
[  3%] Generating moc_euler.cpp
[  3%] Generating moc_library-path.cpp
[  3%] Generating moc_linkage-macros.cpp
[  4%] Generating moc_macros.cpp
[  4%] Building CXX object qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/qxtglobalshortcut.cpp.o
[  4%] Building CXX object api/CMakeFiles/opentrack-api.dir/plugin-api.cpp.o
[  4%] Building CXX object qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/qxtglobalshortcut_mac.cpp.o
[  4%] Generating moc_linkage.cpp
[  5%] Generating moc_math-imports.cpp
[  5%] Building CXX object qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/qxtglobalshortcut_x11.cpp.o
[  5%] Generating moc_math.cpp
[  6%] Building CXX object api/CMakeFiles/opentrack-api.dir/moc_export.cpp.o
[  6%] Building CXX object qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/x11-keymap.cpp.o
[  6%] Linking CXX static library libopentrack-version.a
[  6%] Building CXX object api/CMakeFiles/opentrack-api.dir/moc_plugin-api.cpp.o
[  6%] Generating moc_meta.cpp
[  7%] Generating moc_mutex.cpp
[  7%] Building CXX object api/CMakeFiles/opentrack-api.dir/moc_plugin-support.cpp.o
[  7%] Generating moc_process-list.cpp
[  7%] Building CXX object qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/moc_qxtglobal.cpp.o
[  7%] Building CXX object qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/moc_powerset.cpp.o
[  7%] Building CXX object qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/moc_qxtglobalshortcut.cpp.o
[  7%] Generating moc_qhash.cpp
[  7%] Generating moc_qt-dpi.cpp
[  8%] Building CXX object qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/moc_x11-keymap.cpp.o
[  8%] Building CXX object qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/moc_qxtglobalshortcut_p.cpp.o
[  9%] Generating moc_run-in-thread.cpp
[  9%] Generating moc_qt-signal.cpp
[  9%] Built target opentrack-version
[  9%] Generating moc_shm.cpp
[  9%] Generating moc_sleep.cpp
[  9%] Generating moc_simple-mat.cpp
[  9%] Generating moc_time.cpp
[  9%] Generating moc_sysexits.cpp
[ 10%] Generating moc_thread-name.cpp
[ 10%] Generating moc_tr.cpp
[ 10%] Generating moc_timer.cpp
[ 11%] Building CXX object compat/CMakeFiles/opentrack-compat.dir/base-path.cpp.o
[ 11%] Building CXX object compat/CMakeFiles/opentrack-compat.dir/check-visible.cpp.o
[ 11%] Building CXX object compat/CMakeFiles/opentrack-compat.dir/activation-context.cpp.o
[ 11%] Building CXX object compat/CMakeFiles/opentrack-compat.dir/correlation-calibrator.cpp.o
[ 11%] Building CXX object compat/CMakeFiles/opentrack-compat.dir/camera-names.cpp.o
[ 12%] Building CXX object compat/CMakeFiles/opentrack-compat.dir/euler.cpp.o
[ 12%] Building CXX object compat/CMakeFiles/opentrack-compat.dir/mutex.cpp.o
[ 12%] Building CXX object compat/CMakeFiles/opentrack-compat.dir/qt-signal.cpp.o
[... 70 lines cut ...]
[ 24%] Building CXX object options/CMakeFiles/opentrack-options.dir/globals.cpp.o
[ 25%] Building CXX object options/CMakeFiles/opentrack-options.dir/scoped.cpp.o
[ 25%] Building CXX object options/CMakeFiles/opentrack-options.dir/slider.cpp.o
[ 25%] Building CXX object options/CMakeFiles/opentrack-options.dir/tie.cpp.o
[ 24%] Building CXX object options/CMakeFiles/opentrack-options.dir/group.cpp.o
[ 25%] Building CXX object options/CMakeFiles/opentrack-options.dir/value.cpp.o
[ 25%] Building CXX object options/CMakeFiles/opentrack-options.dir/moc_base-value.cpp.o
[ 25%] Building CXX object options/CMakeFiles/opentrack-options.dir/moc_bundle.cpp.o
[ 26%] Building CXX object options/CMakeFiles/opentrack-options.dir/moc_connector.cpp.o
[ 26%] Building CXX object options/CMakeFiles/opentrack-options.dir/moc_defs.cpp.o
[ 26%] Building CXX object options/CMakeFiles/opentrack-options.dir/moc_export.cpp.o
[ 26%] Building CXX object options/CMakeFiles/opentrack-options.dir/moc_globals.cpp.o
[ 26%] Building CXX object options/CMakeFiles/opentrack-options.dir/moc_group.cpp.o
[ 26%] Building CXX object options/CMakeFiles/opentrack-options.dir/moc_metatype.cpp.o
[ 27%] Building CXX object options/CMakeFiles/opentrack-options.dir/moc_options.cpp.o
[ 27%] Building CXX object options/CMakeFiles/opentrack-options.dir/moc_scoped.cpp.o
[ 27%] Building CXX object options/CMakeFiles/opentrack-options.dir/moc_slider.cpp.o
[ 27%] Building CXX object options/CMakeFiles/opentrack-options.dir/moc_tie.cpp.o
[ 27%] Building CXX object options/CMakeFiles/opentrack-options.dir/moc_value-traits.cpp.o
[ 28%] Building CXX object options/CMakeFiles/opentrack-options.dir/moc_value.cpp.o
[ 28%] Linking CXX shared library opentrack-options.so
[ 28%] Built target opentrack-options
make: *** [Makefile:136: all] Error 2

```

sthalik, 2023-05-27:
> /bin/sh: 1: wineg++: not found

It has to be in PATH.