opentrack/opentrack #1663: Pitching when yawing

jsmith085, 2023-05-27:
Hello OpenTrack community, not sure if this is normal or I'm doing something wrong or misunderstand.  I'm using OpenTrack 2023.1.0 with AITrack 0.6 and things are working ok.  Amazing work by the way.  What's really bugging me about it is that when I turn my head left or right (eg yaw my head), as the raw tracker data yaw value goes above 25 the pitch rapidly increases even though I make a conscious effort to not pitch my head while turning.  It seems as that as the dots marking the jaw line on my face in AITrack pass my mouth and nose it starts pitching more.  The net effect is that in MSFS2020 when I look left or right naturally it pitches and I'm turning down and I see the cockpit not out over the wing.  Is there some way to fix/stop/limit this?

Pieloth, 2023-05-27:
Did you try to use the Neuralnet tracker which is natively proposed in Opentrack? Do you get the same?
Neuralnet tracker does the same as AItrack, but I find it a little bit better, after some tuning setup

erlking, 2023-06-14:
i use Neuralnet since yesterday, so no settings "dialed in" properly, but i have the same issue currently.
it either pitches when yawing or yaws when pitching.

i guess it is happening because the cam is mounted not dead-center because there is the display, so it is a bit above.
But i don't know yet why the calibration does not account that for.

sthalik, 2023-06-14:
It's a bias in the neural network. However I think it's still a lot better than third-party neural networks that connect to opentrack.

bunnitech, 2026-10-15:
Same problem here. I resolved most of it by enabling camera offset. In my case with a cam on top of the monitor I needed a camera offset of pitch 47° roll -2°. I also had to disable neck displacement and fix the head center offset values from "calibration" that put the red dot for rotation center off on my right jawline(?) to get it usable. I'm not sure then that this is actually a neural net bias. Additionally the raw data doesn't show pitch coupling to yaw, it's in the game data output so that's something to do with the handling in opentrack not the model. Check your raw versus output values as you carefully yaw keeping pitch ~constant. If it's making a U shaped motion instead of a horizontal line in the output try messing with the mentioned settings. People keep blaming the model but this case rules that out because it's not in the raw data.