opentrack/opentrack #1668: WinGet incorrectly detects the current version installed

TheSpartaPT, 2023-06-05:
I've initially installed opentrack from the releases page here on github, some time has passed a newer version has been released in the meantime and updated it manually too to `2023.1.0`, just recently i noticed there was a winget package i could be using to make it faster to get it updated, but then noticed it wasn't correctly detecting the current version i have installed, you can see in the package name it says i have `2023.1.0` but the version it detects is `> 2022.3.2` so the package never leaves the `winget update` list

i know if i uninstalled it and reinstalled the latest version it would probably fix itself, but if a newer version is released my issue will probably come back again.

not sure where in the system winget checks which version is installed but if i had to guess it would probably be in 
`HKLM\SOFTWARE\Microsoft\Windows\CurrentVersion\Uninstall\` or `HKCU\SOFTWARE\Microsoft\Windows\CurrentVersion\Uninstall\`

[image]

TheSpartaPT, 2023-06-05:
to sum it up i think this is probably an issue with your installer made with inno setup not updating registry keys correctly when updating

sthalik, 2023-07-23:
Indeed, the installer is flaky like that. It's only going to get fixed by somebody sending a PR for the installer, or having it replaced altogether.