opentrack/opentrack #1669: Neuralnet throwing errors during make on linux

Amordisk, 2023-06-05:
Sorry about opening an issue I just haven't found any way to fix this so far

So I'm trying to build opentrack on ubuntu 20.04 so that it loads the neuralnet module. I've figured out how to get make to start loading the neuralnet module by putting the unzipped onnxruntime into /usr/local and then setting the ONNXRuntime_DIR in ccmake to /usr/local/onnxruntime-linux-x64-1.15.0. The problem is just a little bit after it starts loading the module, it starts throwing errors. I've tried downgrading the onnx version, downloading different dependencies, looking at the code to see if there's anything that's obviously borked (ill admit that part is kinda above me). I've looked at other github issues too but none of those solutions seem to fix it. Here's a picture of the error if it lets me post it.

[image]

If anybody has any type of help I'd greatly appreciate it.

Amordisk, 2023-06-06:
I was downloaded the release version that was packaged before a fix for that specific bug was added. To fix it I git cloned the master branch and followed the normal install process.

closed 2023-06-06 by Amordisk

PLUUZ, 2023-06-08:
hello @Amordisk  I too am having problems building with neuralnet on ubuntu 20.04. I can get it to build without neuralnet.  Can I ask if you're using the ubuntu repo version of opencv? For me the compilation gives an error about the file opencv2/core/quaternion.hpp.  Looking at the opencv website https://docs.opencv.org/4.5.1/db/d65/quaternion_8hpp.html I see that quaternion.hpp does not exist until version 4.5.1  and ubuntu 20.04 has version 4.2 of opencv.

Steps I took: 
$ccmake  . in ./build directory
In ccmake I put in the path to the ONNXRuntime_DIR 

Am I supposed to do something else? I assume the quaternion.hpp error means I need to use a newer version of opencv somehow?
```
[ 81%] Building CXX object tracker-neuralnet/CMakeFiles/opentrack-tracker-neuralnet.dir/deadzone_filter.cpp.o
In file included from ./opentrack-master/tracker-neuralnet/deadzone_filter.h:3,
                 from ./opentrack-master/tracker-neuralnet/deadzone_filter.cpp:1:
../opentrack-master/tracker-neuralnet/unscented_trafo.h:6:10: fatal error: opencv2/core/quaternion.hpp: No such file or directory
    6 | #include <opencv2/core/quaternion.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

Amordisk, 2023-06-08:
I checked my OpenCV version and it seems to be 4.7.0-dev. I don't remember exactly how I installed it but I don't think you installed the correct one, I could be wrong though maybe I just did it weirdly.