opentrack/opentrack #1680: Compile Opentrack with Aruco on Linux

janiolos, 2023-06-29:
I am compiling the Opentrack on Opensuse and I could compile and use it with sucess.
I am trying to include ARUCO on this compiled Opentrack under Linux Opensuse.
what i need to do? need to recompile?
How can I include Aruco in the new compile routine?

sthalik, 2023-06-29:
Use this: https://github.com/opentrack/aruco then set `SDK_ARUCO_LIBPATH`.

VeitHW13, 2023-09-26:
Dear sthalik, could you please be a bit more elaborate about what to do here? I have run into the same issue. I tried to install opentrack with aruco-support following these instructions: https://github.com/opentrack/opentrack/wiki/Building-on-Linux. I finally got opentrack to work, but there was no aruco. I then found this site, on which you gave some advice (using single word answers): https://github.com/opentrack/opentrack/discussions/1435 . However, that did not work opentrack just does not compile. What do I need to do? Do I need to replace the files in the directory tracker-aruco on my PC? Or compile the stuff in that directory with the make files first? Or set the paths differently? I am running Linux Mint and have written software in the past, but I am lost with all these git-ecosystems. Any help would be highly appreciated!

sthalik, 2023-09-26:
Build Aruco from the github.com/opentrack/aruco repository. When building opentrack, set the CMake cache variable `SDK_ARUCO_LIBPATH` to `libaruco.a` from within the aruco build directory. Build opentrack.

VeitHW13, 2023-09-26:
Thanks a lot for this very quick response! However, I ran into a presumably very basic and stupid problem: When I run 'make' I get the utils like 'aruco_creat_marker', but there is no 'libaruco.a' !? Is this hidden someplace or am I missing a command? Here is what I see when running the make-command:

veit@VeitAZi-02:~/Dokumente/2023/kleine_Projekte/OpenTrack/aruco/build$ make
[  9%] Building CXX object src/CMakeFiles/aruco.dir/arucofidmarkers.cpp.o
[ 18%] Building CXX object src/CMakeFiles/aruco.dir/cameraparameters.cpp.o
[ 27%] Building CXX object src/CMakeFiles/aruco.dir/marker.cpp.o
[ 36%] Building CXX object src/CMakeFiles/aruco.dir/markerdetector.cpp.o
[ 45%] Building CXX object src/CMakeFiles/aruco.dir/subpixelcorner.cpp.o
[ 54%] Linking CXX static library libaruco.a
[ 54%] Built target aruco
[ 63%] Building CXX object utils/CMakeFiles/aruco_test.dir/aruco_test.cpp.o
[ 72%] Building CXX object utils/CMakeFiles/aruco_test.dir/cvdrawingutils.cpp.o
[ 81%] Linking CXX executable aruco_test
[ 81%] Built target aruco_test
[ 90%] Building CXX object utils/CMakeFiles/aruco_create_marker.dir/aruco_create_marker.cpp.o
[100%] Linking CXX executable aruco_create_marker
[100%] Built target aruco_create_marker

VeitHW13, 2023-09-26:
ok, I found it - where I never expected it. It is in the src-folder...

closed 2023-11-05 by sthalik