opentrack/opentrack #1684: Hangs on Install Nobara 37

riley-eu-ross, 2023-07-04:
Hi, I'm still pretty new at compiling and installing programs in linux so forgive the novice bumbles please.  CLI still are a little rusty as well.

I'm attempting to install on Nobara 37 and saw #1558. 

When installing the Dependencies I get the following:

```
Problem 1: conflicting requests
  - package wine-devel-8.4-1.fc38.i686 from fedora requires wine-core = 8.4-1.fc38, but none of the providers can be installed
  - package wine-devel-8.6-1.fc38.i686 from updates requires wine-core = 8.6-1.fc38, but none of the providers can be installed
  - package wine-core-8.4-1.fc38.i686 from fedora is filtered out by exclude filtering
  - package wine-core-8.4-1.fc38.x86_64 from fedora is filtered out by exclude filtering
  - package wine-core-8.6-1.fc38.i686 from updates is filtered out by exclude filtering
  - package wine-core-8.6-1.fc38.x86_64 from updates is filtered out by exclude filtering
 Problem 2: conflicting requests
  - package wine-devel-8.4-1.fc38.i686 from fedora requires wine-core = 8.4-1.fc38, but none of the providers can be installed
  - package wine-devel-8.4-1.fc38.x86_64 from fedora requires wine-core = 8.4-1.fc38, but none of the providers can be installed
  - package wine-devel-8.6-1.fc38.i686 from updates requires wine-core = 8.6-1.fc38, but none of the providers can be installed
  - package wine-devel-8.6-1.fc38.x86_64 from updates requires wine-core = 8.6-1.fc38, but none of the providers can be installed
  - package wine-core-8.4-1.fc38.i686 from fedora is filtered out by exclude filtering
  - package wine-core-8.4-1.fc38.x86_64 from fedora is filtered out by exclude filtering
  - package wine-core-8.6-1.fc38.i686 from updates is filtered out by exclude filtering
  - package wine-core-8.6-1.fc38.x86_64 from updates is filtered out by exclude filtering
(try to add '--skip-broken' to skip uninstallable packages)
```

I chose to press on as my understanding is that Nobara has Wine out of the box (sorry I'm kinda new to Wine and how to use it).

This works until I go to the `make` command where I get the following:

`/home/riley/opentrack/options/bundle.hpp:96:44: warning: ‘QMutex::QMutex(RecursionMode)’ is deprecated: Use QRecursiveMutex instead of a recursive QMutex [-Wdeprecated-declarations]
   96 |     QMutex implsgl_mtx { QMutex::Recursive };
      |                                            ^
/usr/include/qt5/QtCore/qmutex.h:159:14: note: declared here
  159 |     explicit QMutex(RecursionMode mode);
      |              ^~~~~~
In file included from /home/riley/opentrack/options/options.hpp:15:
/home/riley/opentrack/options/globals.hpp:20:36: warning: ‘QMutex::QMutex(RecursionMode)’ is deprecated: Use QRecursiveMutex instead of a recursive QMutex [-Wdeprecated-declarations]
   20 |     QMutex mtx { QMutex::Recursive };
      |                                    ^
/usr/include/qt5/QtCore/qmutex.h:159:14: note: declared here
  159 |     explicit QMutex(RecursionMode mode);
      |              ^~~~~~
[ 38%] Building CXX object filter-hamilton/CMakeFiles/opentrack-filter-hamilton.dir/hamilton-tools.cpp.o
[ 38%] Building CXX object filter-hamilton/CMakeFiles/opentrack-filter-hamilton.dir/moc_ftnoir_filter_hamilton.cpp.o
[ 39%] Building CXX object filter-hamilton/CMakeFiles/opentrack-filter-hamilton.dir/moc_hamilton-tools.cpp.o
[ 39%] Linking CXX shared library opentrack-filter-hamilton.so
[ 39%] Built target opentrack-filter-hamilton
[ 39%] Running lupdate for filter-hamilton
[ 39%] Built target i18n-module-filter-hamilton
[ 39%] Generating moc_x11-keymap.cpp
[ 39%] Generating moc_powerset.cpp
[ 39%] Generating moc_qxtglobal.cpp
[ 39%] Generating moc_qxtglobalshortcut.cpp
[ 40%] Generating moc_qxtglobalshortcut_p.cpp
[ 40%] Building CXX object qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/qxtglobalshortcut.cpp.o
[ 40%] Building CXX object qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/qxtglobalshortcut_mac.cpp.o
[ 40%] Building CXX object qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/qxtglobalshortcut_x11.cpp.o
/home/riley/opentrack/qxt-mini/qxtglobalshortcut_x11.cpp:40:10: fatal error: qpa/qplatformnativeinterface.h: No such file or directory
   40 | #include <qpa/qplatformnativeinterface.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/build.make:124: qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/qxtglobalshortcut_x11.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2234: qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
`

Any help would be greatly appreciated.  I have looked around to see if this had happened before but couldn't see anything else.

I also attempted to follow this guide [](https://volatilemark.github.io/opentrack-wine-guide/) and got the same issue.

Installing via snap worked but... it didn't have the plugins that I need for War Thunder and I don't know how to add them that way.

Thanks again and let me know what I can do to help.

sthalik, 2023-07-04:
Install `qt5-qtbase-private-devel`.

Disclaimer: I don't know any .rpm-based distros.

riley-eu-ross, 2023-07-04:
Friendly heads up that I realized I'm actually using Nobara 38 (edited above).

This got me further!  Thank you.  I still have the dep issues and its weird because when I ran the first line in #1558 it included it but I guess it didn't install.

Here is where I am now:
```
In file included from /usr/include/qt5/QtCore/QMutex:1,
                 from /home/riley/opentrack/options/connector.hpp:17,
                 from /home/riley/opentrack/options/bundle.hpp:12:
/usr/include/qt5/QtCore/qmutex.h:159:14: note: declared here
  159 |     explicit QMutex(RecursionMode mode);
      |              ^~~~~~
/home/riley/opentrack/options/bundle.hpp:96:44: warning: ‘QMutex::QMutex(RecursionMode)’ is deprecated: Use QRecursiveMutex instead of a recursive QMutex [-Wdeprecated-declarations]
   96 |     QMutex implsgl_mtx { QMutex::Recursive };
      |                                            ^
/usr/include/qt5/QtCore/qmutex.h:159:14: note: declared here
  159 |     explicit QMutex(RecursionMode mode);
      |              ^~~~~~
In file included from /home/riley/opentrack/options/options.hpp:15:
/home/riley/opentrack/options/globals.hpp:20:36: warning: ‘QMutex::QMutex(RecursionMode)’ is deprecated: Use QRecursiveMutex instead of a recursive QMutex [-Wdeprecated-declarations]
   20 |     QMutex mtx { QMutex::Recursive };
      |                                    ^
/usr/include/qt5/QtCore/qmutex.h:159:14: note: declared here
  159 |     explicit QMutex(RecursionMode mode);
      |              ^~~~~~
[ 68%] Building CXX object proto-udp/CMakeFiles/opentrack-proto-udp.dir/moc_ftnoir_protocol_ftn.cpp.o
[ 68%] Linking CXX shared library opentrack-proto-udp.so
[ 68%] Built target opentrack-proto-udp
[ 69%] Running lupdate for proto-udp
[ 69%] Built target i18n-module-proto-udp
[ 70%] Generating opentrack-wrapper-wine.exe.so
In file included from /home/riley/opentrack/proto-wine/opentrack-wrapper-wine-main.cxx:15:
/home/riley/opentrack/compat/shm.h:10:10: fatal error: windows.h: No such file or directory
   10 | #include <windows.h>
      |          ^~~~~~~~~~~
compilation terminated.
winegcc: /usr/bin/g++ failed
make[2]: *** [proto-wine/CMakeFiles/wine-wrapper.dir/build.make:75: proto-wine/opentrack-wrapper-wine.exe.so] Error 2
make[1]: *** [CMakeFiles/Makefile2:2293: proto-wine/CMakeFiles/wine-wrapper.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
[riley@nobara-pc build]$ 

```
I might see if I can get in touch with https://github.com/tsoulard as well.  I appreciate the timely help.

riley-eu-ross, 2023-07-04:
Well after doing some updating it looks like I have a dependancy issue, the packages hang on update:

```
[riley@nobara-pc build]$ sudo dnf update --nobest
Last metadata expiration check: 0:47:52 ago on Tue 04 Jul 2023 04:54:29 PM EDT.
Dependencies resolved.

 Problem 1: cannot install the best update candidate for package qt5-doctools-5.15.8-2.fc38.x86_64
  - nothing provides libclang.so.16()(64bit) needed by qt5-doctools-5.15.10-1.fc38.x86_64 from updates
  - nothing provides libclang.so.16(LLVM_13)(64bit) needed by qt5-doctools-5.15.10-1.fc38.x86_64 from updates
 Problem 2: package qt5-qttools-devel-5.15.10-1.fc38.x86_64 from updates requires qt5-doctools = 5.15.10-1.fc38, but none of the providers can be installed
  - cannot install the best update candidate for package qt5-qttools-devel-5.15.8-2.fc38.x86_64
  - nothing provides libclang.so.16()(64bit) needed by qt5-doctools-5.15.10-1.fc38.x86_64 from updates
  - nothing provides libclang.so.16(LLVM_13)(64bit) needed by qt5-doctools-5.15.10-1.fc38.x86_64 from updates
 Problem 3: problem with installed package qt5-qttools-devel-5.15.8-2.fc38.x86_64
  - package qt5-qttools-devel-5.15.8-2.fc38.x86_64 from @System requires qt5-designer = 5.15.8-2.fc38, but none of the providers can be installed
  - package qt5-qttools-devel-5.15.8-2.fc38.x86_64 from fedora requires qt5-designer = 5.15.8-2.fc38, but none of the providers can be installed
  - cannot install both qt5-designer-5.15.10-1.fc38.x86_64 from updates and qt5-designer-5.15.8-2.fc38.x86_64 from @System
  - cannot install both qt5-designer-5.15.10-1.fc38.x86_64 from updates and qt5-designer-5.15.8-2.fc38.x86_64 from fedora
  - package qt5-qttools-devel-5.15.10-1.fc38.x86_64 from updates requires qt5-doctools = 5.15.10-1.fc38, but none of the providers can be installed
  - cannot install the best update candidate for package qt5-designer-5.15.8-2.fc38.x86_64
  - nothing provides libclang.so.16()(64bit) needed by qt5-doctools-5.15.10-1.fc38.x86_64 from updates
  - nothing provides libclang.so.16(LLVM_13)(64bit) needed by qt5-doctools-5.15.10-1.fc38.x86_64 from updates
==================================================================================================================================
 Package                             Architecture             Version                             Repository                 Size
==================================================================================================================================
Skipping packages with conflicts:
(add '--best --allowerasing' to command line to force their upgrade):
 qt5-designer                        x86_64                   5.15.10-1.fc38                      updates                   161 k
Skipping packages with broken dependencies:
 qt5-doctools                        x86_64                   5.15.10-1.fc38                      updates                   696 k
 qt5-qttools-devel                   x86_64                   5.15.10-1.fc38                      updates                   227 k

Transaction Summary
==================================================================================================================================
Skip  3 Packages

Nothing to do.
Complete!
```
Let me know if you know of anything to help here, I've reached out to Nobara as well.

tuxtergames, 2023-09-21:
@riley-eu-ross any success? I've the same problem here

jettandres, 2024-02-01:
Hello I also happen to be using Nobara 38. This error only occurred on my end while messing around the setup of onnxruntime/neuralnet tracker

I was able to get this working by recloning the repository and checking out just the `master` branch

afterwards I executed the usual commands outlined in the wiki [for fedora](https://github.com/opentrack/opentrack/wiki/Building-on-Linux#compiling-and-running-opentrack-1) but ran `ccmake .` before `make` and `make install` to configure my onnyx directory

`SDK_WINE` is also turned off for my config. I hope this helps anyone in the future

[image]
[image]