opentrack/opentrack #1689: Higher BaudRate in Hatire Arduino Tracker

GO63-samara, 2023-07-06:
In the `Hatire Arduino` tracker, BaudRate does not exceed `115200` baud.

I transfer additional information between packets in my tracker, so I need higher baud rates supported in the Arduino IDE: `230400`, `250000`, `500000`, `1000000` baud.

Is it possible to introduce these speeds into the Hatire Arduino without changing the Qt sources?

sthalik, 2023-07-07:
See: https://doc.qt.io/qt-6/qserialport.html#BaudRate-enum

Indeed you can try sticking in a non-standard baud rate and see if it works. However, opentrack's user interface doesn't allow you to do it at this time and needs to be modified.

GO63-samara, 2023-07-07:
Yes, I already checked this back in 2020.
Here you can see the list of speeds at 00:08 second: https://youtu.be/NlPE0yR5014

[image]

And it really works, even at `1000000` baud.
But I had to change the `QSerialPort` sources to add the new BaudRates.

I wonder if it's possible to do this using `Opentrack` without touching the `Qt` sources?
For example, write a descendant of the `QSerialPort` class, with a new `enum descendant::BaudRate`.

Or is the only way is to create a PR at https://github.com/qt/qt ?

sthalik, 2023-07-12:
> But I had to change the QSerialPort sources to add the new BaudRates.

What if you make an integer out of the baud rate and cast it to an enum?

GO63-samara, 2023-07-12:
I managed to find the `qserialport.h` file that I corrected.
Looks like all I've changed is:

```
class Q_SERIALPORT_EXPORT QSerialPort : public QIODevice
{
...
public:
...
    enum BaudRate {
        Baud1200 = 1200,
        Baud2400 = 2400,
        Baud4800 = 4800,
        Baud9600 = 9600,
        Baud19200 = 19200,
        Baud38400 = 38400,
        Baud57600 = 57600,
        Baud115200 = 115200,
        Baud230400 = 230400,
        Baud250000 = 250000,
        Baud500000 = 500000,
        Baud1000000 = 1000000,
        UnknownBaud = -1
    };
    Q_ENUM(BaudRate)

```
`enum BaudRate` is in the `public` section.
Maybe it makes sense to make a successor from QSerial?

Right now I won't be able to test your version with an integer and a cast. MSVC is not currently installed.

sthalik, 2023-07-12:
No need to change the file. Just use
```c++
(BaudRate)230400
```
etc.

sthalik, 2023-07-23:
This might be the last thing to do before a release. It's been a while.

closed 2023-11-05 by sthalik

GO63-samara, 2023-11-24:
I discovered that **OT-2023.3.0** introduced "Add non-standard higher baud rates".

I tested it on all speeds from 115200 to 1000000 and it works really well.

Thank you @sthalik !

For Arduino trackers developers it is probably interesting to know the results.
I measured the transmission time of a 30 byte Hatire packet using this code:
```
  unsigned long t0  = micros();

  Serial.write((byte*)&Hat, sizeof(Hat)); // sizeof(Hat) == 30 bytes
  Serial.flush();

  unsigned long t1  = micros();

  Serial.println(" ");  
  Serial.println(t1-t0);  
```

Transfer time 30 bytes for Arduino Nano:
```
  //5108-5112 uS   57600
  //2560-2564 uS  115200
  //1368-1364 uS  230400
  //1208-1212 uS  250000
  // 608- 612 uS  500000
  // 308- 312 uS 1000000
```

By increasing the speed from 115200 to 1000000 you can save 2252 uS for Arduino Nano.

For Arduino Pro Micro, the transfer time of 30 bytes is 28-32 uS, regardless of baudrate.

commit f13d8c9f (2023-11-05): tracker/hatire: add non-standard baud rates

Fixes: #1689