opentrack/opentrack #1693: Debian 12 compilation - libprocps-dev

Vladoslaw, 2023-07-19:
Have just upgraded debian 11 to 12 and wanted to compile a new version of Opentrack in this system. Unfortunetly the libprocps-dev is not avalible in this system. Is there a posibility to jump over this problem? 

Sorry im not a very experienced linux user and i have done only compilation for this aplication, but maby is there a possibility to use the libproc2-dev some how i?

sthalik, 2023-07-20:
Sure, as long as `libproc2-dev` is actually installed. See: https://github.com/opentrack/opentrack/blob/2141b991026f653a9ce343b97400fdb5c4753871/gui/CMakeLists.txt#L14-L19

Vladoslaw, 2023-07-20:
Hi again. Thanks for the answer but i do not know what to do. When I try to  `cmake .` opentrack  i get:
`CMake Error at cmake/opentrack-pkg-config.cmake:14 (message):
  Can't find 'libprocps'.  Please install development files for this package.
Call Stack (most recent call first):
  gui/CMakeLists.txt:14 (otr_pkgconfig)
`
And ibproc2-dev is installed:
`ibproc2-dev is already the newest version (2:4.0.2-3)`

Vladoslaw, 2023-07-20:
Because when i have cp paste your code to the file i had errors i have modified it a bit:

`otr_pkgconfig(${self} libproc2) 

  target_compile_definitions(${self} PRIVATE -DOTR_HAS_LIBPROC2) `

and `cmake .` have pased without errors but durring `make` comand i get error:

`/dane/instalki/opentrack-opentrack-2023.1.0/compat/process-list.hpp:134:10: fatal error: proc/readproc.h: Nie ma takiego pliku ani katalogu
  134 | #include <proc/readproc.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [gui/CMakeFiles/opentrack-user-interface.dir/build.make:207: gui/CMakeFiles/opentrack-user-interface.dir/process_detector.cpp.o] Błąd 1
make[1]: *** [CMakeFiles/Makefile2:1798: gui/CMakeFiles/opentrack-user-interface.dir/all] Błąd 2
make: *** [Makefile:136: all] Błąd 2`

Sorry for my incompetance. I do not know what im doing then it is possible that  I have made something wrong.

"Nie ma takiego pliku ani katalogu" it is "no such file or directory" in English
"Błąd" it is "Error" in English

Any idea how to proceed?

sthalik, 2023-07-20:
```console
$ git describe --tag --always
$ git status
```

Vladoslaw, 2023-07-21:
As always my incompetence have been showed up. When i saw your last answer i have undersand that i should `git clone https://github.com/opentrack/opentrack` the last version and not change anything by my self.
The whole proces went smothly.
So for all Debian 12 and other distros users without `libprocps-dev` you should install `libproc2-dev` instead and follwo the wiki with the istalation.

Dziękuje Panie Stanisławie za pomoc i wyrozumiałość

closed 2023-07-21 by Vladoslaw

clort81, 2023-12-23:
> So for all Debian 12 and other distros users without libprocps-dev you should install libproc2-dev instead and follow the wiki with the installation.

This got me a working opentrack binary. 

PLEASE update the documentation to reflect this.