opentrack/opentrack #1694: Low camera FPS - GUVCVIEW fixes it

domrnet, 2023-07-20:
Hi

I have a strange issue. I have an old logitech c210 usb camera with the ir filter removed to use with 3  IR LED tracker that I made. My camera is getting only 8FPS when I use opentrack however if I open GUVCVIEW and set hardware default and then close it. When I open opentrack again I get a steady 30FPS which is great. Is there some advice on how to get opentrack to stay with 30fps after reboot rather than keep opening GUVCVIEW each time?

GO63-samara, 2023-07-20:
Maybe you need to enable the `MJPEG compression` checkbox?

domrnet, 2023-07-20:
Yep got that on. Solid 8FPS whatever setting. Open guvcview and close it 29-30fps stable in opentrack. I made a small script so when my PC starts up it opens guvcview loads the default then closes guvcview so that is the current workaround.  (when searching the web I stumbled on a video that showed others have an FPS issue on linux and used guvcview to get higher fps https://www.youtube.com/watch?v=iuic3po8kvQ )

Of course the ideal thing will be not to have the workaround and opentrack use my camera right away at 30fps but it's a bit out of my knowledge so any tips welcome :-)

sthalik, 2023-07-21:
Probably has something to do with the exposure setting.