opentrack/opentrack #1695: Compiliation on Steam Deck

WhaleJ84, 2023-07-20:
Has anybody managed to compile opentrack on the Steam Deck? I've been following the instructions on the [Manjaro section of the wiki](https://github.com/opentrack/opentrack/wiki/Building-on-Linux#building-on-manjaro) after running `sudo steamos-readonly disable` but it fails during the `make` phase. I get the following error:

```
/usr/include/c++/12.2.0/x86_64-pc-linux-gnu/bits/os_defines.h:39:10: fatal error: features.h: No such file or directory
   39 | #include <features.h>
      |          ^~~~~~~~~~~~
compilation terminated.
```

Is there some additional packages needed from pacman to get this working? I don't typically use anything Arch based, so I'm not familiar with the architecture.

Saberhawk09, 2023-07-27:
I went down this particular rabbit hole trying to get everything working for head tracking in DCS. Finally decided to take another crack at it, and found this Reddit article to be very helpful.

Basically Valve in their infinite wisdom decided to remove a whole bunch of stuff from SteamOS that is necessary to compile stuff and do basically anything involving "make".

https://www.reddit.com/r/SteamDeck/comments/x4ll4x/install_snap_snapd_zerotier_and_get_aur_working/

Follow everything up to where the guide says to install snapd unless you need that. Keep in mind these commands will reinstall a lot of stuff and might fill your boot / other partitions and make it impossible to install other necessary dependencies and whatnot (speaking from experience). What I did to resolve this was use a bootable Mint USB stick and use gparted to manually resize the partitions that were full.

I hope this helps you! Let me know if you figure out how to compile OpenTrack with the neuralnet tracker included, as that's the step that I got stuck on a few months back.

sthalik, 2023-07-28:
You're meant to use a cross-compiler for steamdeck.

WhaleJ84, 2023-08-12:
> Keep in mind these commands will reinstall a lot of stuff and might fill your boot / other partitions and make it impossible to install other necessary dependencies and whatnot

I don't want to follow any instructions that may cause issues with the Deck, and the Reddit link posted states about the risks of doing so. Ideally we'd want a new section added to the wiki that users can follow without risk (if that is possible).

> You're meant to use a cross-compiler for steamdeck.

Do you know how to do that?

sthalik, 2023-08-12:
>> You're meant to use a cross-compiler for steamdeck.
> Do you know how to do that?

You could probably get away with copying over the Steam Deck base system, chrooting into it and installing headers, then fully bootstrapping a cross-compiler with that system as the *sysroot*. If you ever made a compiler targeting mingw-w64 from Linux, this is the same process.

Or you could even get away with installing a compiler into the copied-over Steam Deck system, to run it inside the chroot as well. Then installing into `/opt/opentrack-version` and treating that as the finished binary.

Otherwise, try simply using `cdebootstrap` with an old `glibc`, then fitting in the same `.so` versions as the SoC device you're targeting, or just bundling them all together with `LD_LIBRARY_PATH`.

The important bits are that glibc isn't forward-compatible (i.e. you cannot compile for one that's newer than the device that you're targeting), as well as `ldd(1)` not returning 'not found' when run on `bin/opentrack` or individual `libexec/opentrack/*.so` files.

WhaleJ84, 2023-08-14:
Hmm, so in reality to get Opentrack working on the Steam Deck isn't feasible for the masses as the majority of people (myself included) does not have the expertise to 'simply' do what you've explained there. But I hope that for the more advanced and dedicated users that we could get some definitive instructions added to the wiki for those seeking it.

Posiris, 2024-10-26:
I was trying to get OpenTrack to work for ATS on Steam Deck, and finally found something that works for me. Idk if it'll work for DCS, but if you are running it through Proton it should. Here is my post on Reddit, with the links to the GitHub. 

https://www.reddit.com/r/linux_gaming/comments/17qsa5d/i_found_out_how_to_get_opentrack_to_work_on_linux/