opentrack/opentrack #1697: opentrack-tracker-kinect-face.dll cannot be found

tmdarlan, 2023-07-25:
getting this error when launching from a batch file: 

DEBUG [F:\dev\opentrack\api/plugin-support.hpp:106]: library "kinect-face" failed: "Cannot load library C:\\Program Files (x86)\\opentrack\\modules\\\\opentrack-tracker-kinect-face.dll: The specified module could not be found."

here is the batch file if that helps.

@echo off
CD "C:\Program Files\Eagle Dynamics\DCS World OpenBeta\bin"
Start "" /B "DCS_updater.exe"
CD "C:\Program Files\Helios Virtual Cockpit\Helios"
Start "" /B "Control Center.exe"
CD "C:\Program Files (x86)\opentrack"
start "" /B "opentrack.exe"
CD "C:\Users\Taylor\Desktop\STUSB"
CALL "libimobiledevice-windows\iproxy.exe" 47047 47047
ping 127.0.0.1 -n 3
Start /B "stusb.exe"
CD "C:\Program Files (x86)\VoiceAttack"
Start "" "voiceattack.exe"
exit

GO63-samara, 2023-07-25:
`kinect-face` tracker does not work under Windows 7. You need Windows 10 or newer.

tmdarlan, 2023-07-25:
Im on windows 11.

GO63-samara, 2023-07-25:
Then just delete the `modules\opentrack-tracker-kinect-face.dll` file.

tmdarlan, 2023-07-25:
that file doesn't exist. that's what's causing the issue right?

sthalik, 2023-07-26:
The file `C:\Program Files (x86)\opentrack\modules\\opentrack-tracker-kinect-face.dll` clearly exists. You can also ignore this warning.

closed 2023-11-05 by sthalik