opentrack/opentrack #1701: neuralnet tracker not building by default?

enjoycowboy, 2023-08-05:
(Arch linux)
Repo is cloned as normal. When cmake runs, no mention of neuralnet tracker...
```
-- The C compiler identification is GNU 13.2.1
-- The CXX compiler identification is GNU 13.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- User settings file '/home/enjoy/opentrack/sdk-paths-enjoy@GNU-Linux.cmake' doesn't exist
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.8.1") 
-- module api
-- module compat
-- module csv
-- module cv
-- module filter-accela
-- module filter-ewma
-- module filter-hamilton
-- module filter-kalman
-- module filter-nm
-- module user-interface
-- module logic
-- module migration
-- module executable
-- module options
-- module pose-widget
-- module proto-fgfs
-- module proto-libevdev
-- module proto-udp
-- module qxt-mini
-- module spline
-- module tracker-easy
-- module tracker-freepie-udp
-- module tracker-fusion
-- module tracker-linux-joystick
-- module tracker-pt-base
-- module tracker-pt
-- module tracker-s2bot
-- module tracker-test
-- module tracker-udp
-- module video-opencv
-- module video
```

Same happens when installing from AUR, either the opentrack or opentrack-git packages.

sthalik, 2023-08-05:
Install `onnxruntime-bin`.

enjoycowboy, 2023-08-05:
> Install `onnxruntime-bin`.

hmm, weird... i had `onnxruntime` built from the `onnxruntime-git` package. It was able to build correctly with `onnxruntime-git `but with `onnxruntime-bin` it doesnt build.

```
ninja: build stopped: subcommand failed.
==> ERROR: A failure occurred in build().
    Aborting...
 -> error making: opentrack-exit status 4
 -> Failed to install the following packages. Manual intervention is required:
opentrack - exit status 4

```

sthalik, 2023-08-06:
If you can build `onnruntime-git` that's good enough for specifying `ONNXRuntime_DIR`.

closed 2023-11-05 by sthalik