opentrack/opentrack #1707: settings dialog and .ini bugs and glitches

sthalik, 2023-08-27:
If you know a bug or a glitch related to how configurable settings are processed (number fields, selections from list, etc.), please report them. An example of a bug is if a settings gets applied despite pressing 'cancel'.

Here's some things that must always work.

For all things except very heavy ones (camera name or camera fps, etc.), they are applied immediately. So if you set camera FOV without pressing OK, it already took effect by the time you write it in.

But if you press 'cancel', the settings must revert itself to the one from where you pressed OK previously or had it in the config file. Similarly, cancelled settings cannot under any circumstance save themselves into the .ini.

Similarly, if you change the used .ini with the dialog open and/or the tracker running, it must take effect immediately. If you press 'cancel' by then, the version from the newly-selected .ini must take effect (without clobbering it with the temporary one).

Please report all bugs like this (or freezes/crashes related to settings windows).

Len62, 2023-08-29:
The key bindings default to ctrl+, even though ctrl is not pressed. They work, but if you revert back to 2023.1.0, they have to be re-set. I'll with a clean ini file to see if it happens still.
[image]

sthalik, 2023-08-29:
@Len62, I'm positive there was no change to shortcut and options handling in 2023.2.0.

Len62, 2023-08-29:
Ok, thank you. It only happens when I use a copied profile. I'll try a few more and see what happens.

evilphish, 2024-01-16:
How about the Camera Settings button on the neuralnet tracker not doing anything when clicked? Also the FPS seems to be locked to 29 even though the camera can do 60.
[image]