opentrack/opentrack #1714: Can't using HTC Tracker to Steam VR

hanmike, 2023-09-10:
[image]

sthalik, 2023-09-10:
Is this an OpenVR SDK version issue? Right now opentrack is built against OpenVR SDK 1.3.22 version.

Are any other users having that problem?

I'm going to try updating it but it's going to take a few days.

hanmike, 2023-09-11:
I solved it by reinstalling(SteamVR,opentrack) and restarting(computer) a few times.

Below is a description of the problem situation.

When Steam VR is run as administrator, if just run opentrack, nothing appears as shown in the picture. When run as administrator, the program stops the moment press the input settings button, after about 5-10 minutes, it works again, but can't see anything, just like in the picture.

-Google translation
[image]