opentrack/opentrack #1720: It doesn't working in the background

stradalee, 2023-09-16:
If I minimize opentracker window after running it, it suddenly disappears from the taskbar.

So if I go into the task manager and search opentracker, it's running now, but it doesn't use hardware such as cpu, and it doesn't work normally when I run it again. So there is a hassle of having to force to shut it down and run it again.

There was no such problem in previous versions.

sthalik, 2023-09-19:
I won't be able to fix it without additional information or a way to reproduce it. Worse, there haven't really been any changes in 2023.2.0 that could cause this. It might be difficult to diagnose and fix.

sthalik, 2023-09-19:
Are you sure you haven't enabled tray support?

stradalee, 2023-09-20:
Ahhh, I haven't checked that yet.

stradalee, 2023-09-20:
Disabling tray support, all issues disappeared. 
But there is new question. Why doesn't opentrack work on dcs if I enable tray support?

sthalik, 2023-09-20:
Tray support shouldn't change anything regarding freetrack output.

stradalee, 2023-09-20:
I see. Thank you for your help.

closed 2023-09-20 by sthalik