opentrack/opentrack #1721: Linux Upgrade 3.9.0.6 to 3.10.0.10 database fails upgrade so no logins.

Freejack01, 2023-09-17:
Attempting upgrade and database fails so no logins.

[09-17-2023 @ 20:43:30.151 GMT] [startUpgrade()@H2Store2.java:170] java.nio.file.NoSuchFileException: ./dbbackup/phantombot.h2.mv.db
[09-17-2023 @ 20:43:30.154 GMT] [startUpgrade()@H2Store2.java:170]
java.nio.file.NoSuchFileException: ./dbbackup/phantombot.h2.mv.db
        at java.base/sun.nio.fs.UnixException.translateToIOException(Unknown Source)
        at java.base/sun.nio.fs.UnixException.rethrowAsIOException(Unknown Source)
        at java.base/sun.nio.fs.UnixException.rethrowAsIOException(Unknown Source)
        at java.base/sun.nio.fs.UnixCopyFile.copyFile(Unknown Source)
        at java.base/sun.nio.fs.UnixCopyFile.copy(Unknown Source)
        at java.base/sun.nio.fs.UnixFileSystemProvider.copy(Unknown Source)
        at java.base/java.nio.file.Files.copy(Unknown Source)
        at com.gmt2001.datastore2.H2Store2.startUpgrade(H2Store2.java:167)
        at com.gmt2001.datastore2.H2Store2.<init>(H2Store2.java:95)
        at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
        at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(Unknown Source)
        at java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(Unknown Source)
        at java.base/java.lang.reflect.Constructor.newInstanceWithCaller(Unknown Source)
        at java.base/java.lang.reflect.Constructor.newInstance(Unknown Source)
        at com.gmt2001.datastore2.Datastore2.init(Datastore2.java:226)
        at com.gmt2001.datastore2.Datastore2.init(Datastore2.java:150)
        at tv.phantombot.PhantomBot.<init>(PhantomBot.java:336)
        at tv.phantombot.PhantomBot.main(PhantomBot.java:1182)

I tried both the full version and the lin version and same result.

sthalik, 2023-09-18:
Wrong repo.

closed 2023-09-18 by sthalik

Freejack01, 2023-09-18:
Yep realized it after you replied, sorry..... only issue with opentrack on Linux is onnxruntime compile and when run with opentrack it seg faults.

sthalik, 2023-09-19:
Send a backtrace.