opentrack/opentrack #1722: Issues using Easy Tracker 1.1

117-Yash, 2023-09-21:
Hello guys, I'm using the Kinect V2 camera with Easy Tracker 1.1 for DCS. however Even after I made my own clip and bought a store bought TrackIR replica clip, I am having issues with it. The QI keeps jumping and it doesn't seem to get a steady hold of the positions. Even when i remove my glasses to prevent any extra IR readings (it usually picks up my eyes as seen), it still has issues holding position. Can I have some help please?
[image]

Slion, 2023-09-21:
As it is it does not support clip markers as mentioned there, only hats I'm afraid:
https://github.com/opentrack/opentrack/wiki/Easy-Tracker

117-Yash, 2023-09-22:
Oh, the "Model" section in the settings had a clip selection so I assumed, thank you.

closed 2023-09-22 by 117-Yash

Slion, 2023-09-22:
> Oh, the "Model" section in the settings had a clip selection so I assumed, thank you.

Interesting, I'm guessing I left it there ready to be implemented. I'm pretty sure I never actually did it. I don't even have a clip 😁

117-Yash, 2023-09-23:
> > Oh, the "Model" section in the settings had a clip selection so I assumed, thank you.
> 
> Interesting, I'm guessing I left it there ready to be implemented. I'm pretty sure I never actually did it. I don't even have a clip 😁

Hey Slion, after your comment, I spent a time making a hat tracker and I'm still facing some issues with the smoothness and transition. 

This is a long shot as I understand that you have a life and can't do this for everyone, but may I request some help with setting mine up? Maybe over a 30 min call on a platform of your preference? 

I understand if you say no, I get it however I'm just a student with not much experience and it would be of tremendous help to me if you could do so. Thanking you in advance.

Kind regards,
Yash

Slion, 2023-09-23:
Of course, ping me on discord mate:
https://discord.com/invite/7M4Ms5dMZE