opentrack/opentrack #1734: Aruco has a roll problem

ddzzkunye, 2023-10-01:
> Aruco has a similar problem  

_Originally posted by @ddzzkunye in https://github.com/opentrack/opentrack/issues/1731#issuecomment-1741886358_

ddzzkunye, 2023-10-01:
[image]  
I _drew_ this ArUco.

ddzzkunye, 2023-10-01:
[image]
I drew this ArUco.

sthalik, 2023-10-02:
The PT and Aruco trackers didn't have these problems back when they were being developed.  Maybe try it with a known good marker. Here are some [user-submitted videos](https://www.dropbox.com/sh/atyazm22wwuzxlf/AADOStjt0F-xIsmhOcbEKXyZa?dl=0).

If there's a global issue then it needs to be fixed properly. There shouldn't be a need to invert any output axes.

ddzzkunye, 2023-10-03:
In all the videos I could find, the aruco is rotated 90 degrees.  
[image]
[image]

sthalik, 2023-10-04:
> Maybe this historical problem shouldn't be fixed? It affects too many people.

You're right that the problem exists, but it's mostly of theoretical interest. Users are expected to wave it at the Aruco tracker before mounting the marker on their headsets.

closed 2023-10-21 by ddzzkunye