opentrack/opentrack #1736: Logitech Brio 500

Gyoniro, 2023-10-07:
Ticking/Unticking the MJPEG box doesn't produce any noticeable effect in neuralnet.. Can you also put in a resolution textbox? Because this Webcam needs to be set to 1280x720 to produce 60FPS...

_Originally posted by @GBry in https://github.com/opentrack/opentrack/issues/1421#issuecomment-1025134303_
            
Can't seem to find a solution to run the webcam at 60fps inside of opentrack. If I try to lower the resolution to ease the load, the webcam defaults at ~30fps.

muhammadnaufal72, 2024-09-06:
Running into this issue as well, 60fps seem to work fine on other software

nigel100100, 2025-01-24:
Same issue here. Even with lowest exposure and low light compenstation off it will not ever run higher than 30fps in Opentrack.

adameric1119s, 2026-01-31:
Seems like HDR locks it to 30fps, disabling it in Logitech GHub fixed the issue for me, running at stable 60fps at 720p

[image]