opentrack/opentrack #1739: Mouse emulation not working on some games

Snowblind58, 2023-10-12:
Hello,

Recently have been using the mouse emulation output to assist alongside my gaming endeavours.
When I lauch "7days to die" and "unturned" the mouse emulation works on the UI game menus but when Im in game it doesn't move mc Chars head.

Please Help me as I am Sad.
Kind regards
Officer Smith.

sthalik, 2023-11-05:
Did you try both DInput and Raw modes? Are you launching the game or opentrack as administrator? Try launching both as admin or neither.

GO63-samara, 2023-11-05:
Maybe the game is loading NPclient.dll and is therefore not responding to the mouse?
Need to clear a key in the registry?

nscreated, 2023-11-23:
I would like to add Fallout 4 to the list of games that don't work with mouse emulation. Neither DInput nor Raw modes work.

I'm not a developer but I've done some digging and here's what I think. I think that OpenTrack might not be respecting "DInput" and might be treating all mouse emulation as "Raw". That's because its current behavior is identical to the way Moonlight+Sunshine behaves when the user chooses "Optimize mouse for remote desktop instead of games" (which causes them to prefer absolute positioning of the mouse instead of relative positioning). Absolute positioning is bad for games. It will work in their menus but not in the game itself.

OpenTrack has code that uses different mouse input methods depending on whether the user selected "DInput" or "Legacy" with Legacy using SetCursorPos(), which is absolute, and DInput using SendInput(), which is relative.

SendInput() should work. Therefore, I would check to make sure the code is definitely using SendInput() when it's supposed to.

I am not a developer, so everything I wrote is probably poop.

sthalik, 2023-11-23:
@nscreated, see https://github.com/opentrack/opentrack/blob/master/proto-mouse/ftnoir_protocol_mouse.cpp#L55-L76

The raw input mode uses [SendInput](https://learn.microsoft.com/en-us/windows/win32/api/winuser/nf-winuser-sendinput) with the `MOUSEEVENTF_MOVE` flag, while the legacy mode uses [SetCursorPos](https://learn.microsoft.com/en-us/windows/win32/api/winuser/nf-winuser-setcursorpos).

sthalik, 2023-11-23:
> SendInput() should work. Therefore, I would check to make sure the code is definitely using SendInput() when it's supposed to.

You can probably use `procmon.exe` to trace it. Otherwise you connect a debugger after downloading the `dbginfo` package from the releases page.

nscreated, 2023-11-23:
I tried procmon.exe and monitored opentrack.exe while filtering out the UDP Receive Operations and Registry operations. Mostly I got Process Profiling operations which tended to list the active opentrack DLLs at each moment, but otherwise I didn't see anything that would give me clues about exactly how opentrack was moving the mouse.

Guess I will try the debugger route.

nscreated, 2023-11-25:
OK! I used the debugger. And I can confirm that OpenTrack is using SendInput() just as it is supposed to. So my hypothesis is wrong.

So why is SendInput() not being respected in some games? It's a headscratcher.

sthalik, 2023-11-25:
No idea, but if you know of any other input methods then be my guest.

nscreated, 2023-11-25:
Bing Chat suggests mouse_event(). [https://sl.bing.net/kL8MWimeyLk](https://sl.bing.net/kL8MWimeyLk)

Only compatible with Windows though.

Snowblind58, 2023-11-25:
> No idea, but if you know of any other input methods then be my guest.

I think it’s games that are made in unity, certain games just don’t do it and it’s mainly games from unity.

GO63-samara, 2023-11-25:
Have you checked whether the key is in the registry?:
`HKCU\SOFTWARE\NaturalPoint\NATURALPOINT\NPClient`

When emulating a mouse, it will not work. You need to remove this key, like here:
https://github.com/opentrack/opentrack/issues/1312#issuecomment-901805347

nscreated, 2023-11-25:
> Have you checked whether the key is in the registry?: `HKCU\SOFTWARE\NaturalPoint\NATURALPOINT\NPClient`
> 
> When emulating a mouse, it will not work. You need to remove this key, like here: [#1312 (comment)](https://github.com/opentrack/opentrack/issues/1312#issuecomment-901805347)

Tried it, no change. 

> > No idea, but if you know of any other input methods then be my guest.
> 
> I think it’s games that are made in unity, certain games just don’t do it and it’s mainly games from unity.

Fallout 4 is Creation Engine.

nscreated, 2023-11-26:
Update.

The problem isn't with SendInput() as such. It should work.

For example, here is Sunshine's Windows implementation of SendInput() and it works just fine. 
https://github.com/LizardByte/Sunshine/blob/336062d467d5f48ba56d05a523c212f791421108/src/platform/windows/input.cpp#L442

sthalik, 2023-11-26:
What's the simplest game to set up for testing? Preferably one that doesn't *need* to be paid for.

nscreated, 2023-11-26:
@Snowblind58 says it fails on Unity games. Here's a list of Unity games on Steam with many of them free:

https://store.steampowered.com/curator/39750107-Games-Made-With-Unity/#browse

nscreated, 2023-11-27:
I've tested it on the following games, all of which work fine:

Fallout 3
Skyrim
Hades
Witcher 3
Disco Elysium (Unity)
Ori and the Blind Forest (Unity)

So the problem is not related to game engines I don't think. Fallout 4 is the only Bethesda game that has the problem and the only game I've tested so far with the issue.

The only thing Fallout 4 has in common with OP's games (7 Days to Die and Unturned) is that they are all FPSes.

nscreated, 2023-11-29:
Evidently, this is a very common issue for people developing alternative input strategies for games.

https://www.google.com/search?q=sendinput+games+c%2B%2B&sca_esv=586305682&sxsrf=AM9HkKnScN_guE77LMA4S8YKLyS9bA-jbg%3A1701269393851&ei=kU9nZcG4M4nqptQPheK9-AU&oq=sendinput+games+c%2B%2B&gs_lp=EhNtb2JpbGUtZ3dzLXdpei1zZXJwIhNzZW5kaW5wdXQgZ2FtZXMgYysrMgoQABhHGNYEGLADMgoQABhHGNYEGLADMgoQABhHGNYEGLADMgoQABhHGNYEGLADMgoQABhHGNYEGLADMgoQABhHGNYEGLADMgoQABhHGNYEGLADMgoQABhHGNYEGLADSPw3UABYAHACeAGQAQCYAQCgAQCqAQC4AQPIAQDiAwQYACBBiAYBkAYI&sclient=mobile-gws-wiz-serp

Sropen, 2024-07-07:
Same issue here. Mouse emulation don't work on Ravenfield (unity game).