opentrack/opentrack #1745: No connection over UDP after moving

QuackerCracker, 2023-11-06:


QuackerCracker, 2023-11-06:
I've been using open track with a head tracking app on my phone with pretty good results for months, though I recently moved (twice) and now it's no longer connecting. The first time everything was fine, I simply changed the ipv4 address and opentrack/app worked perfectly together. Though now that I'm in a my new place I can't connect via UDP anymore. I tried pretty much every address possible, restarted my computer, reinstalled opentrack and even tested a different app though nothing seems to work. For clarification I used opentrack/app together maybe 2 days ago, unplugged all of my stuff to move my PC, and plugged it in today. I haven't installed anything, moved files, etc. since using opentrack last and now it's not working.

sthalik, 2023-11-06:
Try using Wireshark.

QuackerCracker, 2023-11-06:
> Try using Wireshark.

I know you can monitor incoming/outgoing connections through wireshark though I've never used it myself. Did do a quick test and there's definitely data going through the address, however it doesn't seem like there's any going from my phone to the computer, at least I'm assuming. 

I know the IP of my phone on the wifi [ex. .11] and my computer [ex. .12], though I don't see them communicating at all. The Source bracket does not have my phone [.11] sending data to my computer [.12] in the Destinations bracket. The port is definitely open, as I made sure of that in my firewall and turned off the firewall briefly to test.