opentrack/opentrack #1748: global shortcuts do not work anymore (without sudo)

erlking, 2023-11-07:
Hi,

if i run opentrack it accepts the keybind i set in the options but the program does not actually accept the input.
so if i set "Center" to e.g. ALT+Q and "toggle tracking" to ALT+W, i confirm with OK, i press the keybinds even while having the application in focus, nothing happens.
if i start opentrack with sudo (and confirm the "do not run the application with root" messages, the global shortcuts work.

 > WARN [/var/home/martin/Downloads/tools/opentrack-opentrack-2023.3.0/qxt-mini/qxtglobalshortcut.cpp:97]: QxtGlobalShortcut failed to register: QKeySequence("Alt+Q")
> WARN Requested decoration  "adwaita"  not found, falling back to default
> WARN Wayland does not support QWindow::requestActivate()
> WARN Wayland does not support QWindow::requestActivate()
> WARN Requested decoration  "adwaita"  not found, falling back to default
> WARN Wayland does not support QWindow::requestActivate()
> WARN [/var/home/martin/Downloads/tools/opentrack-opentrack-2023.3.0/qxt-mini/qxtglobalshortcut.cpp:97]: QxtGlobalShortcut failed to register: QKeySequence("Alt+W")
> WARN [/var/home/martin/Downloads/tools/opentrack-opentrack-2023.3.0/qxt-mini/qxtglobalshortcut.cpp:97]: QxtGlobalShortcut failed to register: QKeySequence("Alt+E")
> WARN [/var/home/martin/Downloads/tools/opentrack-opentrack-2023.3.0/qxt-mini/qxtglobalshortcut.cpp:97]: QxtGlobalShortcut failed to register: QKeySequence("Alt+W")
> WARN [/var/home/martin/Downloads/tools/opentrack-opentrack-2023.3.0/qxt-mini/qxtglobalshortcut.cpp:97]: QxtGlobalShortcut failed to register: QKeySequence("Alt+E")
> ^C
> ⬢[martin@toolbox ~]$ sudo /var/home/martin/Downloads/tools/opentrack-opentrack-2023.3.0/build/install/bin/opentrack
> QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'

any help getting it to work would be appreciated.
thanks

on fedora 38 silverblue, with wayland.
i usually run it inside a toolbox, but the same issue appears if i run it on my system
today i compiled 2023.3 and the problem also exists in this version (used 2023.1 until now)

sthalik, 2023-11-07:
How about having opentrack receive shortcut-like events on the command-line? Would that work better for you on Wayland?

erlking, 2023-11-07:
yeah, would be great in the future!

i found a workaround in another issue regarding global shortcuts which works (for now?)  "opentrack -platform xcb"
so i guess i can close it :)

thanks

closed 2023-11-07 by erlking