opentrack/opentrack #1752: Recommended settings for the PS3 Eye camera

microraptor, 2023-11-10:
It would be very useful to add recommended settings to the end of the [PS3 Eye camera driver wiki article](https://github.com/opentrack/opentrack/wiki/PS3-Eye-open-driver-instructions) for the point tracker, neuralnet tracker and the Aruco tracker.

Currently at the end of that page is a screenshot which shows 320x240 @ 200Hz being selected and MJPEG unchecked. Are these recommended settings? According to the [PS3 Eye Wikipedia article](https://en.wikipedia.org/wiki/PlayStation_Eye) 187 FPS is the maximum. It also sounds like MJPEG is generally a good thing and should be enabled.

[This Wiki page](https://github.com/opentrack/opentrack/wiki/PT-tracker-stability-settings#ps3-eye-camera-settings) explicitly mentions not to use 320x240 or 640x480@75Hz for the point tracker.

Most comments I found seem to use for resolution 640x480 and for FPS 60 or 75. However, they are mostly about the point tracker.

So in the past 640x480@60Hz seems to be the most recommended option for the point tracker. Does this change with the new open-source driver and the MJPEG option? Perhaps the neuralnet or Aruco runs better with lower resolution and higher FPS settings. Did someone do any testing on this?

Only the FOV setting seems to be certain:
The field of view (FOV) depends on which zoom level the lens barrel is rotated at. On the very left side (red dot) the FOV is 56 and on the very right side (blue dot) the FOV is 75.

beggehorn, 2023-12-01:
I just picked up a PS3 Eye camera yesterday and have some of the same questions. I'm using neuralnet and can only select 30/60 FPS in that mode. Would like to get the FPS higher if possible.

beggehorn, 2023-12-14:
The neuralnet settings only allow 30 or 60. My settings are 640x480 @ 60 but OpenTrack is only showing a maximum of 45 FPS.