opentrack/opentrack #1753: Wiki page about AI face tracking

microraptor, 2023-11-11:
AI face tracking became very good and popular in recent years. However, there doesn't seem to be any information about these trackers on the wiki. I wrote a short summary as seen below, which can be posted on the wiki, but I don't have any editing rights. It might be better to split this into three separate articles. Any corrections and additions are very welcome.

---

# AI Face Tracking

The AI models used for face trackers made a lot of progress and became very good in recent years. There are reports of people, who already own dedicated hardware such as TrackIR, but use Opentrack and an AI face tracker instead, because they don't have to wear any special equipment on their head and it performs such as well for them.

However, if face tracking doesn't work reliably for you, you can also try the [Aruco tracker](https://github.com/opentrack/opentrack/wiki/Aruco-tracker). If you want more reliability (especially in dark environments), use a setup using the point tracker and an IR clip, like the ones sold by [Delanclip](https://delanclip.com) or [Grass Monkey Simulations](https://grassmonkeysimulations.com).

## List of AI face trackers with Opentrack integration

Neuralnet
- free and open-source software
- included in Opentrack, no additional installation necessary
- uses a normal webcam or PS3 Eye camera for tracking
- there are reports that it performs better than the older AITrack model
- more info: https://github.com/opentrack/neuralnet-tracker-traincode

AITrack
- free and open-source software
- uses a normal webcam or PS3 Eye camera for tracking
- more info and installation: https://github.com/AIRLegend/aitrack

Eyeware Beam
- paid and proprietary software
- uses a normal webcam for tracking
- besides head tracking, it also does eye tracking and combines both
- alternatively they also offer an iOS app
- more info and installation: https://beam.eyeware.tech/
- [see the wiki for the iOS app](https://github.com/opentrack/opentrack/wiki/An-iPhone-iPad-Head-&-Eye-Tracker-w-OpenTrack)

SmoothTrack
- paid and proprietary software
- runs on iOS and Android devices and uses their native AR capabilities 
- uses the device camera for tracking and sends data over the network to opentrack
- can be used with USB cable instead of WiFi to reduce latency
- more info and installation: https://smoothtrack.app/

Tobii Eye Tracker
- can be used with Opentrack or its own proprietary software
- uses a proprietary device as camera
- there are reports it has slightly higher latency than other trackers
- besides headtracking it also provides eye tracking
- more info: https://gaming.tobii.com/

## Webcams for tracking

Although most software using webcams work with pretty much any webcam, high refresh rate (60+ FPS) webcams usually perform better tracking. These can be quite expensive. Therefore it is widely recommended to use a PlayStation 3 Eye camera, which are are specifically made for tracking (they have a very high refresh rate on low resolutions) and are commonly sold on Ebay used for about 10 USD/EUR. [An open-source driver for the PS3 Eye camera](https://github.com/opentrack/opentrack/wiki/PS3-Eye-open-driver-instructions) is released by Opentrack.

For the PS3 Eye camera the most recommended settings are 640x480@60Hz with MJPEG enabled. The field of view (FOV) depends on which zoom level the lens barrel is rotated at. On the very left side (red dot) the FOV is 56 and on the very right side (blue dot) the FOV is 75.

Apps like [DroidCam](https://www.dev47apps.com/) can also be used to use a phone or tablet as a webcam. It is recommended to connect the phone to the PC via USB instead of using WiFi for lower latency.

## Settings

Although it is recommended to tweak your settings manually to achieve the best performance, here is a list of guides and settings for orientation.

General:
- https://github.com/opentrack/opentrack/wiki/Accela-in-opentrack-2.3
- https://github.com/opentrack/opentrack/wiki/Quick-Start-Guide-(WIP)
- https://forum.il2sturmovik.com/topic/34403-a-complete-guide-to-set-up-head-tracking-opentrack/
- https://github.com/AIRLegend/aitrack/wiki/Common-camera-FOV-values

Neuralnet tracker:
- Delanclip publishes preconfigured config files in their manuals (change camera settings if you don't use the DelanCam1):
https://delanclip.com/manuals/face-tracking-setup/

Neuralnet tracker with Logitech C920:
- https://www.digitalcombatsimulator.com/en/files/3331374/
- https://imgur.com/gallery/QT86LBf

Airtrack tracker:
- https://github.com/AIRLegend/aitrack/wiki/User-guide
- https://old.reddit.com/r/StarWarsSquadrons/comments/kbpq2s/use_opentrack_aitrack_to_get_headtracking_with/

Smoothtrack:
- https://smoothtrack.app/instructions.html (At then end of the instructions a config file with curves is provided)

sthalik, 2023-11-12:
It's definitely better if you can edit the page by yourself. I sent you an invitation.

microraptor, 2023-11-12:
Thank you, I joined the organization. However, I don't have permissions to create or edit wiki pages yet. I assume you have to assign them manually. Let me know if I should the webcam and settings section on separate wiki pages, instead of leaving them as one.

I will then link the face tracking page at the one [FAQ question](https://github.com/opentrack/opentrack/wiki/common-issues#i-dont-want-to-make-a-led-clip) and in the paragraph where the [Quick Start Guide](https://github.com/opentrack/opentrack/wiki/Quick-Start-Guide-(WIP)) talks about trackers.

sthalik, 2023-11-12:
It seems the only way to grant you wiki edit rights is to grant you de facto repository write access. I trust you won't do anything bad with it.

As for clip manufacturers, there's also [TrackHat](https://www.trackhat.org/) in the UK.

microraptor, 2023-11-12:
Thank you, I added trackhat, cleaned up the links and created [the wiki page](https://github.com/opentrack/opentrack/wiki/AI-Face-Tracking). I also added the links in the quick start guide and faq:
[image]
[image]

microraptor, 2023-11-12:
I saw the issue with people posting malware links when the wiki was public. In the long run it would probably best to keep the documentation/wiki as files on the main or a separate repository and have a github action/CI [update the github wiki](https://stackoverflow.com/a/58692638) or create a web page (like Docusaurus, MkDocs, GitBook or Read the Docs). Then people could submit PR requests for wiki changes.

closed 2023-11-12 by microraptor

sthalik, 2023-11-12:
That's a good idea. A simple Github Action can execute a force-push into the hidden wiki repository.