opentrack/opentrack #1755: Neuralnet not creating TrackIR data

Teriy-Aki, 2023-11-14:
The neuralnet tracker initially works fine and reads my face movement successfully, creating coordinates from it. Still, Opentrack does not seem to create a TrackIR dummy file in Freetrack Output (no matter the settings) - TrackIR-dependent tracking does not translate to games as a result.

[image]

sthalik, 2025-05-01:
NPClient doesn't create files. It creates a named shared memory mapping that isn't accessible by browsing files.

Make sure that you've enabled TrackIR output in freetrack 2.0 options.

AirPix26, 2025-05-01:
Thank you, I will try your recommendation.