opentrack/opentrack #1758: Opentrack won't open with Ferdora 39

martinpetrus, 2023-11-19:
After build from `master`, opentrack fails to start with following error:
```
$ opentrack
opentrack: error while loading shared libraries: libprocps.so.8: cannot open shared object file: No such file or directory
```

```
$ uname -r
6.5.11-300.fc39.x86_64
```

This is due to procps-ng refactoring and the renaming of the above mentioned library ([link](https://gitlab.com/procps-ng/procps/-/releases/v4.0.0)).

"This project contains builds of the newest release of procps-ng. The version procps-ng-4.0.2 contains a new, massively rewritten library. This library changed its filename from /usr/lib/libprocps.so to /usr/lib/libproc2.so ([link](https://discussion.fedoraproject.org/t/jrybar-procps-ng-4-0-2-rebase/46822))."

This will likely be an issue as these changes start to make it into further mainstream distros.

sthalik, 2023-11-19:
See https://github.com/opentrack/opentrack/blob/6bdc009eced70fcfdb78a4f3905481950f1099ef/gui/CMakeLists.txt#L14C1-L19

Add development headers for `libproc2` for cmake to find it.

martinpetrus, 2023-11-19:
Thank you for the swift reply!

I'm still at a loss. I have devel packages installed:
```
$ sudo dnf install *proc2*
Last metadata expiration check: 0:28:29 ago on Sun 19 Nov 2023 09:55:50 PM CET.
Package procps-ng-4.0.3-5.fc39.x86_64 is already installed.
Package procps-ng-4.0.3-5.fc39.i686 is already installed.
Package procps-ng-devel-4.0.3-5.fc39.x86_64 is already installed.
Package procps-ng-devel-4.0.3-5.fc39.i686 is already installed.
Dependencies resolved.
Nothing to do.
Complete!
```
`pkg-config` clearly sees the lib:
```
$ pkg-config --path libproc2
/usr/lib64/pkgconfig/libproc2.pc
```
Header files are in place:
```
$ ll /usr/include/libproc2/
total 88
-rw-r--r--. 1 root root  4559 Feb 13  2023 diskstats.h
-rw-r--r--. 1 root root 10100 Feb 13  2023 meminfo.h
-rw-r--r--. 1 root root  2943 Feb 13  2023 misc.h
-rw-r--r--. 1 root root 14042 Feb 13  2023 pids.h
-rw-r--r--. 1 root root  4881 Feb 13  2023 slabinfo.h
-rw-r--r--. 1 root root  6197 Feb 13  2023 stat.h
-rw-r--r--. 1 root root 24135 Feb 13  2023 vmstat.h
-rw-r--r--. 1 root root  6689 Feb  7  2023 xtra-procps-debug.h
```

I wonder why `otr_pkgconfig_` would still return `false`, but I haven't looked into the code itself and presume I'm still doing something wrong...

sthalik, 2023-11-19:
If you only recently installed `libproc2.pc`, the CMake cache could be out of date. Try recreating the build directory.

martinpetrus, 2023-11-19:
Works now. Thank you! The lib likely sneaked in with a system update. I had had cleaned the build directory previously, but the caches slipped my attention.

```
$ git status
On branch master
Your branch is up to date with 'origin/master'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	tracker-rs/rs_impl/CMakeCache.txt
	tracker-rs/rs_impl/CMakeFiles/
	tracker-rs/rs_impl/Makefile
	tracker-rs/rs_impl/cmake_install.cmake
```
After deleting these, all went fine.
I much appreciate your insight and assistance!
I will close the ticket now.

closed 2023-11-19 by martinpetrus