opentrack/opentrack #1759: easier building of opentrack

sthalik, 2023-11-23:
After switching to Qt 6, we can make opentrack build easily using submodules. It won't require installing Qt or any other third-party dependencies.

Right now, https://github.com/opentrack/opentrack-depends requires manual setup. But once the core opentrack code can be built easily, there's finally a point in making it work using CMake submodules (see my https://github.com/sthalik/floormat/blob/master/external/CMakeLists.txt soluton for another project).  

One issue is proprietary non-redistributable dependencies. However, they can be added to private repos. CI can be set up to fetch private repos in read-only mode. Project developers also can gain access to non-redistributable dependencies since it's used within an organization, not given out to other/random entities.

sthalik, 2023-11-23:
Also how do you feel about switching away from Qt? Maybe it's time for opentrack v3 with a different GUI, removal of `compat/simple-mat` etc.

sthalik, 2023-11-23:
Pinging the usual suspects: @GO63-samara, @DaWelter, @Slion.

Some ideas;

Replace `options::value<T>` / `options::tie_setting` with this system. I'm starting to hate `QSettings`, `QVariant` and such. The [new system](https://github.com/sthalik/floormat/blob/2611db2/editor/inspect-types.cpp#L25-L65) is more declarable. And yeah, replace all Qt with a declarative + RAII-based MVVC framework that enables a web + mobile UIs without sacrificing the desktop experience for existing users (as an example of a well-made GUI + Web app, see qbittorrent).

Replace `QMutexLocker` with a custom semi-userspace mutex (futex, waitforaddress, pthread spinlock + mutex + condition variable).

sthalik, 2023-11-25:
For project developers, check out the https://github.com/opentrack/depends-nonfree repository. This should help you a bit setting up your sdk paths.

sthalik, 2023-11-25:
How does everyone feel about switching to Qt 6 within the next few months?

GO63-samara, 2023-11-25:
Qt6 and Qt5 can both be installed at the same time, but in different folders, right?
Then everything is fine.

Cat-Lady, 2024-01-28:
While considering path to future, please consider less barebone (not to say utterly terrible) linux build/installation support. It is 2024, things like Steam Deck exist, and - honestly - much more complicated FOSS software have much more sane release pipeline on the OS for... a decade, or more.

Not a personal problem for me - I compiled it successfully with all addons I needed, but I must say it was most unfriendly project (counting ones that do not have technical reasons to not work) to *ever* compile with the needed functionality. I guess it hinders adoption quite much...