opentrack/opentrack #1763: Opentrack very not translating movement to input

raidr11, 2023-11-30:
**Description of Issue**
I haven't been using my OpenTrack for a couple of weeks, and when I started it up today, it didn't work. 
I can see the dots from my delan clip move around in the window to the left, but it’s not translated to tracker data and game data and the octopus doesn't move, or move erratically.

**Steps taken**
I've tried to update to the latest version of Opentrack. I\ve updated my win 11 to the latest version.
I've experimented with different framerate settings, input and output - all to no avail. 
I've tried to run it as administrator

**Screenshots**
Enclosed are screenshots of OpenTrack , CL Eye Test and Windows version in a PDF Document.

[OpenTrack_Movement-not-translated-to-game.pdf](https://github.com/opentrack/opentrack/files/13514635/OpenTrack_Movement-not-translated-to-game.pdf)

sthalik, 2023-11-30:
Your point extraction threshold is too high given that two points have merged into one. So decrease it.

raidr11, 2023-11-30:
thank you for your swift reply, it worked like a charm