opentrack/opentrack #1768: Ps3 Eye driver problem.

giuse26, 2023-12-07:
Hello !!! I´m trying to setup ps3eye tracking with opentrack for il-2.
The problem is that after installing the drivers. I get the "Can´t open camera. PS3 Eye driver error:Unknown error" message.
If I set the input to "Easy tracker 1.1" I get the same message but there is an image feed in the program.
I´ve installed the drivers as instructed by the wiki.

Things that I´ve tried:
- Different PS3 Eye camera
- Different USB ports / hubs
- Uninstalling and reinstalling the drivers 

The PC is running windows 11 PRO 22H2.
This is a picture of the Device manager

[image]

I´ll really appreciatre your help. 
Please forgive any errors. I´m not a native english speaker

sthalik, 2023-12-07:
Maybe try switching between libusb0 and libusbK drivers for the device.

giuse26, 2023-12-07:
> Maybe try switching between libusb0 and libusbK drivers for the device.

Tried it ... didn´t work :(

sthalik, 2023-12-07:
Sorry, I can't think of anything else right now. Maybe somebody else will know.

giuse26, 2023-12-07:
https://github.com/opentrack/opentrack/issues/1495#issuecomment-1772758557
  that´s it ... problem solved. Thanks!!!!

closed 2023-12-07 by giuse26

ericrisler, 2024-11-09:
> [#1495 (comment)](https://github.com/opentrack/opentrack/issues/1495#issuecomment-1772758557) that´s it ... problem solved. Thanks!!!!

This worked for me as well. Very strange