opentrack/opentrack #1771: Options window too large, apply button can't be seen is appears to still be an issue.

diemonstar, 2023-12-10:
I believe I am using the latest version of opentrack (2023.3.0) I'm on Windows 11 Pro with my desktop set to 4k 3840x2160 120Hz. I've tried several of Windows' scaling options but opentrack appears to ignore all of them. I am using a 65" LG OLED C2 as my monitor.
[image]

sthalik, 2023-12-10:
Try setting the environment variable `QT_SCALE_FACTOR` to `1`, or `1.25`, `1.5`, `1.75` and so on. It this doesn't help, try setting `QT_FONT_DPI` to `96`, `120` or `144`.

diemonstar, 2023-12-10:
I didn't figure out how to use QT_SCALE_FACTOR. I get an error in a command line console or powershell when I use it. I think I'm missing something. But when searching how to use that method to fix the scaling issue I found a workaround creating and executing a .vbs script. Now I have it properly sized when starting via the script. Thank you for pointing me mostly in the right direction. :)