opentrack/opentrack #1777: Is there a way to auto-start tracking when starting a game from the "game detection list"?

djesse, 2023-12-21:
Hi,

Is there a way (besides having to press a hotkey) to automatically start tracking when an app from my "game detection" list is starting. Switching profiles works like charm but I still have to manually press "Start" to activate tracking. If not, it would be a great addition as I use my PC for work and gaming. So I don't need it to track me all the time, but when I startup a game, this would things make a little smoother.

Thanks.

Len62, 2023-12-22:
It works like that for me. I guess just check your settings in the game detection tab:
[image]

djesse, 2023-12-22:
Maybe, you missunderstood my issue. Its not that the profiles don't change. Its the tracking itself that I would like to start automatically.

[image]

Len62, 2023-12-22:
But it does start automatically. With OT running, but not tracking, if I run either of those games in the list, tracking starts. And when I exit the game tracking stops.

djesse, 2023-12-22:
It doesn't do that here. It switches the profiles between DCS, StarCitizen or MSFS but usually I have to start tracking manually. And  I would expect to have a checkbox to enable/disable automatic tracking to start/stop.

Len62, 2023-12-22:
Ok, first, make sure the "start game profiles..." box is checked. Second, are you running OT, and games, with Admin Privileges?

djesse, 2023-12-22:
[image]

The "start game profiles ..." is checked. OpenTrack is starting with Windows. So its always there. My account has admin priviliges, but the games are started without that "Start as Administrator" checkmark in their file config.

sthalik, 2023-12-22:
I don't know how this is possible. See: https://github.com/opentrack/opentrack/blob/master/opentrack/main-window.cpp#L964-L968

djesse, 2023-12-23:
You're right. I checked on my second monitor ... tracking DOES indeed start. It's an issue on my side - for MSFS I have to use one of the starter-helper exe files to trigger the profile change. While this helper Exe is run when I start MSFS, it probably gets shut down early and triggers the stop of the tracking.

The reason why I use the helper EXE is, because I use the MS Store version of MSFS. Files on my harddrive seem to be encrypted or have some other "special protection". Anyway this has the effect that I can't choose the correct "EXE" file in your file request when connecting the profile. There comes an error message that I dont have the priviliges to run this exe (even Admin is not enough). As OT needs the path and filename, but hides the path, I can't simply enter the full location (path+filename) into the list.

You can close the issue. Maybe this explanation helps to find a better solution for the game detection list in such edge cases.

sthalik, 2023-12-23:
Try just typing 'FlightSimulator.exe' in the game detector's executable name.