opentrack/opentrack #1779: Webcams cannot be detected in all opentrack inputs?

GauRugeila, 2023-12-23:
Hi, I've used AITrack+opentrack with my logitech C270 no problem, but when I tried to use aruco, neuralnet, etc. the camera name just leave blank, and no reaction when I click camera settings. I've tried all usb ports and reinstall everything. The weirdest is, I'm using a laptop and even my built-in cam also cannot be detected.
[image]

sthalik, 2023-12-25:
See if the same issue occurs in opentrack 2023.1.0 and 2023.2.0.

GauRugeila, 2023-12-26:
> See if the same issue occurs in opentrack 2023.1.0 and 2023.2.0.

2.0 not working, then I tried "opentrack-test-20230527_01.7z" version which solves the issue, maybe my laptop or windows 10 version is too old. Thanks.

sthalik, 2024-01-22:
See if this build works for you: https://ananke.misaki.pl/opentrack/opentrack-test-20240121_01.7z

GO63-samara, 2024-01-22:
I checked in Windows7, in `OT-test-20240121_01` the list of video cameras is still empty.
In addition, the `neuralnet` tracker disappeared from the list.
In `OT-2023.3.0-win32-portable` the list of video cameras is empty, but the `neuralnet` tracker is in the list.

```
C:\Users\Win7>D:\OT-test-20240121_01\opentrack.exe

C:\Users\Win7>
DEBUG [F:\dev\opentrack\api/plugin-support.hpp:106]: library "kinect-face" faile
d: "Cannot load library D:\\OT-test-20240121_01\\modules\\\\opentrack-tracker-ki
nect-face.dll: DEBUG [F:\dev\opentrack\api/plugin-support.hpp:106]: library "neu
ralnet" failed: "Cannot load library D:\\OT-test-20240121_01\\modules\\\\opentra
ck-tracker-neuralnet.dll: DEBUG [F:\dev\opentrack\api/plugin-support.hpp:106]: l
ibrary "win32-mouse" failed: "Cannot load library D:\\OT-test-20240121_01\\modul
es\\\\opentrack-proto-win32-mouse.dll: DEBUG [F:\dev\opentrack\api/plugin-suppor
t.hpp:106]: library "opencv" failed: "Cannot load library D:\\OT-test-20240121_0
1\\modules\\\\opentrack-video-opencv.dll: DEBUG [F:\dev\opentrack\api/plugin-sup
port.hpp:106]: library "ps3eye" failed: "Cannot load library D:\\OT-test-2024012
1_01\\modules\\\\opentrack-video-ps3eye.dll:
```

sthalik, 2024-01-24:
Is there any specific compiler and Qt version that should be used in order to support Windows 7?

GO63-samara, 2024-01-26:
Compiler and Qt versions have nothing to do with it, because I checked your finished assembly:
https://ananke.misaki.pl/opentrack/opentrack-test-20240121_01.7z

Tested on Windows 7 SP1 64bit. The problems started with version 2023.2.0. In version 2023.1.0 everything works well.
I suspect the problems are due to a fix in version 2023.2.0:
`video/opencv: apply bugfixes for Windows 11 and enable Media Foundation input.`

I also checked `opentrack-test-20240121_01` on Windows 10. The list of cameras is displayed correctly.

Perhaps Opentrack is no longer fully supported in Windows 7? Perhaps it's time for everyone to switch to Windows 10/11?

sthalik, 2024-02-11:
I'm still trying things out to debug the Windows 7 situation. It's possible that it's a static initializer again that was only exposed after a compiler upgrade— much like the one successfully debugged and fixed by DaWelter.

GO63-samara, 2024-02-11:
> I also should get to optimizing cv::floodFill(). The PT tracker spends some 10-15% of all time there.

Yes! I remember this: https://github.com/opentrack/opentrack/issues/1651

sthalik, 2024-02-12:
Alright I posted the build in https://github.com/opentrack/opentrack/issues/1394#issuecomment-1938864870

GO63-samara, 2024-02-12:
It is excellent!

Under Windows 7 in version `opentrack-test-20240212_01` there are 19 trackers in the list of trackers. All trackers except `Kinect-face`.
All connected video cameras are also in the list.
Everything is working.
I didn't even have to install `VC_redist.x86`.

Thank you, @sthalik !