opentrack/opentrack #1780: Opentrack not recognising my camera

SeaMine12, 2023-12-25:
Around last week I was using opentrack and it worked as normal but now I have reinstalled windows and I am installing opentrack 2023 3.0 again though when I press the start button it just shows nothing as well as there being no tracking done. I found a post from 2019 of a user having the same issue and the fix for the issue on the post was to go to your profile directory and delete all the ini files but that doesn't work either for me.
[image]

sthalik, 2023-12-25:
See if the same issue occurs in opentrack 2023.1.0 and 2023.2.0.

SeaMine12, 2023-12-26:
Alright

SeaMine12, 2023-12-26:
Opentrack 2023 1.0 works though it doesn't have nuralnet tracker. I'll just use AI track

boPD, 2024-01-19:
So is this issue solved? My camera is black too but I dont know where to download 2023 1.0 or 2.0

moe-sc, 2024-03-22:
> So is this issue solved? My camera is black too but I dont know where to download 2023 1.0 or 2.0

Check the [releases page](https://github.com/opentrack/opentrack/releases).

boPD, 2024-03-22:
邮件已收到